Marus/cortex-debug #198: Ports already in use [remote container]

Keriz, 2019-08-27:
Hello,

I'm trying to automate my toolchain installation for the STM32 and want to have debugging capabilities included. I am using  the vs-code remote container feature and installed Cortex-Debug as a remote extension on the container side. However when I try to debug (F5 key), an error window pops up and says 

[image]

Are there any ports that I need to expose in docker ? I already exposed the port 4444 for openocd.

Any help would be greatly appreciated! Thanks.

Keriz, 2019-08-28:
Excellent! Somehow I didn't see this issue. 
I ran into another issue right after this one:

```
Open On-Chip Debugger 0.10.0+dev-g0819541 (2019-08-28-09:01)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J21S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.277385
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 50000 for gdb connections

```
I went throught other issues on github and it seems that some people had the same problem (https://github.com/Marus/cortex-debug/issues/166). 
The best option would be to change my openocb build?

Keriz, 2019-08-28:
I think I already build the last one. Here is how I get it and build it in the Dockerfile. (there are tools required that I fetch with apt-get before)

```
RUN cd /usr/src/ \
&& git clone --depth 1 https://github.com/ntfreak/openocd.git \
&& cd openocd \
&& ./bootstrap \
&& ./configure --enable-stlink --enable-jlink --enable-ftdi --enable-cmsis-dap \
&& make -j"$(nproc)" \
&& make install
```

I tried your command. Here is the output:
```
Open On-Chip Debugger 0.10.0+dev-g0819541 (2019-08-28-09:01)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J21S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.280587
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
```

Keriz, 2019-08-28:
I just double check that the necessary openocd config was good, everything seems OK from that side. Also, I was able to flash the device before if this gives any information.

```
Please check OUTPUT tab (Adapter Output) for log of /usr/local/bin/openocd
Launching server: "/usr/local/bin/openocd" "-c" "gdb_port 50000" "-s" "/workspaces/oibus-mini-adc-dac/freertos-firmware" "-f" "/usr/local/share/openocd/scripts/interface/stlink.cfg" "-f" "/usr/local/share/openocd/scripts/target/stm32f3x.cfg"
1-gdb-set target-async on
2-interpreter-exec console "source /root/.vscode-server/extensions/marus25.cortex-debug-0.3.2/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor reset halt"
5-target-download
6-interpreter-exec console "monitor reset halt"
7-enable-pretty-printing
```
The output tab now displays `Info : Listening on port 50000 for gdb connections` if I run openocd from the command line

I also see `./arm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory` when I try to manually start arm-none-eabi-gdb. I'm having a look into that right now. But maybe the extension runs its own version of gdb?

Keriz, 2019-08-28:
Yep, I configured my settings.json as such:

`{
    "telemetry.enableTelemetry": false,
    "cortex-debug.armToolchainPath": "/root/opt/gcc-arm-none-eabi-8-2019-q3-update/bin/",
    "cortex-debug.openocdPath": "/usr/local/bin/openocd"
}`

Anything

Keriz, 2019-08-28:
Somehow basing my docker on a previous ubuntu and installing libncurses5-dev made it work! and now everything runs smoothly. Your help has been very useful, thank you very much.

Last thing maybe, do you know when you'll roll out the next extension version ? It would be smoother than having to download the file on the other issue. Thanks again!

Marus, 2019-09-04:
Sorry - back at this now. Will be looking at @haneefdm's change and be trying to get them out in the near future. Sorry for the delays.

Travis-Spidertracks, 2019-09-10:
We are using the custom build from Haneefdm which allows the use case described in the issue - very useful feature. Thanks for making it available, and it will be great to have this in a formal release.

haneefdm, 2020-02-11:
I think this issue is resolved and should be closed

closed 2020-02-11 by haneefdm