Marus/cortex-debug #199: 【warning】Couldn't determine a path for the index cache directory.

webees, 2019-08-29:
How to solve this warning? Please help me, thank you.

[image]

# version
- openocd-20190715.7z
- gcc-arm-none-eabi-8-2018-q4-major-win32.exe
- VSCode-win32-x64-1.37.1.zip

# vscode extensions
- Cortex-Debug 0.3.1
- ARM 0.5.0
- C/C++ 0.25.0

# console
```
undefinedD:\@ARM_TOOLS\8-2018-q4-major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
```

# launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "cortex-debug",
      "request": "launch",
      "preLaunchTask": "build",
      "name": "Debug (OpenOCD)",
      "servertype": "openocd",
      "cwd": "${workspaceFolder}",
      "executable": "${workspaceFolder}/boot.elf",
      "configFiles": [
        "${workspaceFolder}/openocd.cfg"
      ]
    }
  ]
}
```

# c_cpp_properties.json
```
{
  "configurations": [
    {
      "name": "STM32",
      "cStandard": "c11",
      "cppStandard": "c++17",
      "intelliSenseMode": "clang-x64",
      "compilerPath": "arm-none-eabi-gcc",
      "defines": [
        "STM32F4"
      ],
      "includePath": [
        "${workspaceFolder}",
        "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include",
        "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1",
        "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1/arm-none-eabi",
        "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1/backward",
        "D:/@ARM_TOOLS/8-2018-q4-major/lib/gcc/arm-none-eabi/8.2.1/include",
        "D:/@ARM_TOOLS/8-2018-q4-major/lib/gcc/arm-none-eabi/8.2.1/include-fixed"
      ],
      "browse": {
        "path": [
          "${workspaceFolder}",
          "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include",
          "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1",
          "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1/arm-none-eabi",
          "D:/@ARM_TOOLS/8-2018-q4-major/arm-none-eabi/include/c++/8.2.1/backward",
          "D:/@ARM_TOOLS/8-2018-q4-major/lib/gcc/arm-none-eabi/8.2.1/include",
          "D:/@ARM_TOOLS/8-2018-q4-major/lib/gcc/arm-none-eabi/8.2.1/include-fixed"
        ],
        "limitSymbolsToIncludedHeaders": false,
        "databaseFilename": ""
      }
    }
  ],
  "version": 4
}
```

haneefdm, 2019-08-29:
You are getting that warning from gdb. Probably harmless but setting the environment variable HOME to the same value as USERPROFILE may make gdb happy. But this can affect other Unix-y tools as well, mostly in a good way, but just letting you know of potential side effects. You can also turn off index caching to disk or change where gdb wants to store it in your .gdbinit in your home directory which again probably relies on your HOME env. variable.

https://sourceware.org/gdb/onlinedocs/gdb/Index-Files.html

closed 2019-08-30 by webees

hwmaier, 2021-07-21:
Just a note: setting the `XDG_CACHE_HOME` environment variable to something like `c:\User\myname\.cache` worked for me.