Marus/cortex-debug #200: Setup verification check

matszal, 2019-08-30:
HI Marus and other contributors.

I have installed recently your plugin and followed the steps from SEGGER and web but I have a couple questions in terms of configuring the ports and speed of the JLink probe. 
How to change the default port numbers and speed of the device which is set by default to 4kHz and 50000...1,2?

Below are my config.json files and debug terminal. For some reason the code goes into exception. Again I am new to gdb debugging. Perhaps you could point me out if my configuration is correct?

      `{
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe",            
            "cwd": "${workspaceRoot}",
            "executable": "${cwd}/iron-sparrow/app/sumx.elf",
            "name": "Debug (J-Link)",
            "device": "cortex-a7",  
            "interface": "jtag",        
        
            "showDevDebugOutput": true,
            "runToMain": true
        }`

`Please check OUTPUT tab (Adapter Output) for log of C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe
Launching server: "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe" "-if" "jtag" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "cortex-a7"
1-gdb-set target-async on
2-interpreter-exec console "source c:/Users/SESA548172/.vscode/extensions/marus25.cortex-debug-0.3.1/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
8-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from C:\\dev/iron-sparrow/app/sumx.elf..."}]}
Reading symbols from C:\dev/iron-sparrow/app/sumx.elf...
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"done.\n"}]}
done.
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"wdog_trigger () at ../aos/aos/wdog.c:102\n"}]}
wdog_trigger () at ../aos/aos/wdog.c:102
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"102\t    for (;;);\n"}]}
102	    for (;;);
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["frame",[["addr","0x801b41be"],["func","wdog_trigger"],["args",[]]]],["file","../aos/aos/wdog.c"],["fullname","C:\\dev\\iron-sparrow\\aos\\aos\\wdog.c"],["line","102"]]}]}
Not implemented stop reason (assuming exception): undefined
GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"connected","results":[]}}
GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
9-thread-list-ids
10-thread-list-ids
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
`

haneefdm, 2019-08-30:
If you are thinking of stuff like gdb/telnet/swo TCP ports, these are chosen for you. Stuff like `interface`, `device`, `serialNumber`, `ipAddress`. `rtos`, `jlinkscript` can be set in `launch.json`.

About speed, you have a few options. JLink supports setting certain things like speed using gdb commands.
```javascript
"preLaunchCommands": [
    // Set JTAG speed to 30 kHz
    "monitor speed 30"
],
"postLaunchCommands": []. // in case you need to use this
"postRestartCommands": [], // doubt you would need this but its there.
```
See the JLink gdb server [user guide](http://www.cs.tut.fi/~sulo/harjoitustyo/ymparisto/JLinkGDBServer.pdf) See section 3.5

You can also put some commands in a jlink script file.

matszal, 2019-09-02:
@haneefdm many thanks for quick reply and help on this. I was able to change the speed but have no control of changing the port number from 50000 to 2331 for example. I think that's not the big issue. The main problem is, I can't see the code that I want to debug on VSCODE window after flashing the target MCU. I am working on a windows 10 machine. Is this setup enough and sufficient to achieve that? Thanks guys in advance for helping me out!

Screen with json setup and  debug console attached. 
[image]

haneefdm, 2019-09-02:
It appears to be working? It stopped at the reset/exception vector and the source code seems to be in 'exception.S'. Click on the stack trace in the 'Call Stack' window. It should take you to line 59.

Since you enabled 'runToMain' in launch.json, it should have continued to your main program. Did it not?

Btw, are you starting the gdb server yourself? Then you would have two copies running because this extension starts one automatically -- the command-line version. The two copies then may fight for control over the HW access. Check the 'Output' tab to the left of 'Debug Console' In the 'Output' tab, select the 'Adapter Output' from the drop-down list on the right. You will see the output from  jlink. It has been a while since I used JLink myself.

Yes, you cannot change the port number fro 50000,1,2 This extension automatically finds a free port number(s) if those ports are busy.

If you HAVE to start the gdb-server manually you can but with a different configuration and it may be a bit awkward.

matszal, 2019-09-02:
> It appears to be working? It stopped at the reset/exception vector and the source code seems to be in 'exception.S'. Click on the stack trace in the 'Call Stack' window. It should take you to line 59.
>
Yes, that is working fine and takes me to line 59 as you said. Thank you ;)
> Since you enabled 'runToMain' in launch.json, it should have continued to your main program. Did it not?
> 
Unfortunately that does not bring me to main at all? Seems to be stuck at Starting target CPU...?

> Btw, are you starting the gdb server yourself? Then you would have two copies running because this extension starts one automatically -- the command-line version. The two copies then may fight for [...]
So, I specified path to JLink GUI instead of CL, that's why I was able to see the interface. I changed path to /SEGGER/JLink/JLinkGDBServerCL.exe, and now can see "Adaptor Output" - thank you again!

> Yes, you cannot change the port number fro 50000,1,2 This extension automatically finds a free port number(s) if those ports are busy.
> 
> If you HAVE to start the gdb-server manually you can but with a different configuration and it may be a bit awkward.

That fully answer my question regarding port config.

Would be great if you could further collaborate with me about my main.c issue or point me out in right direction whenever suits please.

haneefdm, 2019-09-02:
I will take a look at the 'runToMain' issue. In your 'Debug Console' you should see a '-exec-continue' gdb command which I don't see. Can you save the output in that window into a text file and attach it here?

matszal, 2019-09-02:
I spotted "-exec-continue" there, please see log attached below
[output.txt](https://github.com/Marus/cortex-debug/files/3566301/output.txt)

haneefdm, 2019-09-02:
That looks like the 'exec-continue' worked. Command 13 was setting the breakpoint and command 16 was a continue. So, it should have worked. Command 17 failed but is not relevant since the target is running.
```
13-break-insert -t --function main
GDB -> App: {"token":13,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","1"],["type","breakpoint"],["disp","del"],["enabled","y"],["addr","0x801be4d8"],["func","main"],["file","../aos/!hw05/main.c"],["fullname","C:\\dev\\iron-sparrow\\aos\\!hw05\\main.c"],["line","151"],["thread-groups",["i1"]],["times","0"],["original-location","-function main"]]]]}}
...
16-exec-continue
GDB -> App: {"token":16,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
```
At this point, I suspect an issue in your pre-main startup code. Disable `runToMain` and try single stepping from your reset vector? It appears that the program exeution never reaches main after that continue. You may have had a hardware exception and then in an infinite loop.

matszal, 2019-09-02:
Let me check everything again and I will report, once I know more

matszal, 2019-09-06:
@haneefdm Hi, we had some issues with disabling the watchdog. That's why the code wasn't working when trying to debug directly from command line and gdb. The exception issue happened because in the command line gdb was exectued with following arguments `arm-linux-musleabihf-gdb -q -tui -iex "target remote :2331" -x sumx.gdb`. The sumx.gdb is a configuration file that sets the the memory map from other .elf file before executing the second .elf file. My last question is is there a way to tell vscode launch.json to run that config file which will load both .elf files?

haneefdm, 2019-09-06:
Yes, in launch.json you can specify `debuggerArgs` to pass along to gdb.

matszal, 2019-09-06:
Thanks but im still having issues parsing "debuggerArgs". Here's the part of the gdb file:
[SAmpleGDB.txt](https://github.com/Marus/cortex-debug/files/3582987/SAmpleGDB.txt) If you could  please just hint me a proper way of lading them args, it would be fantastic!

haneefdm, 2019-09-06:
Yes, that gdb command file will probably confuse this extension because there is a `continue` in it. Could you give me the whole launch.json to see what else you got? Perhaps I can replicate and see if I can find you a workaround. I can see there are two elf files involved but I want to know the rest.

If you are having trouble with the `launch.json` syntax?
```javascript
"debuggerArgs" = [
    "-x",
    "sumx.gdb"
],
```

You can also use the `.gdbinit` file in the current working dir.

matszal, 2019-09-06:
My launch.json file here: 
[launchJSON.txt](https://github.com/Marus/cortex-debug/files/3584207/launchJSON.txt)

spachner, 2020-12-31:
Hi,

regarding using jlink from within VS Code: Does anybody know how to set the address to which the binary is being flashed with jlink? Defaults to 0x0, but would like to use a different.

regards

spachner

haneefdm, 2021-01-02:
VSCode/Cortex-Debug does not control the address. Your executable does via your linker script. We just hand the executable (the elf file) to gdb to let it do its thing

gdb on the other hand relies on the JLink gdb server to do its thing. For a launch (not attach), Cortex-Debug does issue a reset (via gdb) but it is left to the gdb server to do its thing

spachner, 2021-01-02:
Hi,

thanks for answering. Sure, the linker script determines the address the binary is linked to. I used to have a bootloader binary and an application binary which sits one after another in flash. The bootloader is linked to 0x0000 and app to e.g. 0xB000. I am coming from Eclipse CDT which allows me to flash only the application, remaining the bootloader in flash untouched. This is quite handy for debugging the application which is startet by the bootloader. Under Eclipse when starting the JLink debugger I have to pass the right start address the application binary is linked to and to this address the flashing has to take place, too. 

Attached is a sceenshot from Eclipse. 
[image]

Hoped anybody does know how to get the same accomplished with vscode, too.

regards

spachner

RisinT96, 2021-01-02:
You'll have to modify your linkerscript to link it to a different base address, also perhaps modify your makefile to build two separate ELFs, one with (or maybe only) and one without a bootloader, then modify your launch json to have separate launch configurations for bootloader and firmware debug, each using the appropriate elf.

I don't see any other way with vscode.

My team simply compiles both the bootloader and firmware into a single elf that's then debugged together, but if we just want the firmware without a bootloader we build a slightly different version that has the startup code and is linked to the "original" base address.

spachner, 2021-01-02:
Hi RisinT96,

exactly, all what you describe are we already doing. The only thing I am missing is the option for the launch.json how to set the address the binary is being flashed to. We do not want to compine elf files.

Currently we use this:
```
        {
            "name": "Debug J-Link Tag -- flash",
            "request": "launch",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/tag/build/last_build.elf",
            "servertype": "jlink",
            "serverpath": "${config:knx_launch_serverpath}",
            "armToolchainPath": "${config:knx_launch_armToolchainPath}",
            "serialNumber": "", //If you have more than one J-Link probe, add the serial number here.
            "rtos": "FreeRTOS",
            "interface": "swd",
            "runToMain": true,
            "device": "ATSAMD21G18",
            "svdFile": "${workspaceRoot}/tools/segger/ozone/svd/ATSAMD21G18A.svd",
            "flash_start_address": 0xB000",   // <<<<< what is the option for something like this.
            "preLaunchCommands": [
                "monitor speed 6000",
                "monitor flash_start_address 0xB000",   // <<<<< what is the option for something like this.
            ],                           
        },

```

RisinT96, 2021-01-02:
If your linkerscript is correct, then your firmware in `last_build.elf` should already be linked to `0xB000`, and launching debug with that elf would program the firmware to that address, as that's what gdb does.

The only problem is you'd need to have your bootloader already pre-programmed from beforehand.

spachner, 2021-01-02:
Ha, this is great, it simply works! I did not expect this, since under Eclipse we have to supply the right start flash address. So I didnt't tried it under vscode and kepts on searching a while for the start_address option...

Many thanks!

By the way, we did not alter the linker file, instead we pass an option to linker in Makefile as linker option:

`--section-start=.text=$(TEXT_START_ADDR)`

regards

spachner

RisinT96, 2021-01-02:
Glad that worked for you!

Though I think your solution of directly modifying the `.text` section's position can be problematic, it should be fine most of the time as most sections in flash are placed after `.text` relatively.
But there may be some additional sections that come before the `.text` section, for example in the linkerscript supplied by ST, the vector table is in a separate section that comes before `.text`, your solution will leave the vector table at `0x0000`, which is quite problematic.

You gotta make sure that everything is placed relatively to `0xB000`, and that your bootloader updates the `VTOR` register.

spachner, 2021-01-02:
👍

PhilippHaefele, 2021-12-26:
@haneefdm I think we can close this one

The initial issues from matszal has been resolved here. Just the special case (unrelated to cortex debug) with using `debuggerArgs` for disableing the watchdog wasn't completely resolved. I will try to create an example in the wiki for using a `.gdbinit` file with a hint to not use continue in it.
@matszal If there are any open questions, please try with newest version of Cortex Debug and create a new issue for that

2nd (mixed in) issue from spachner seems to be resolved

closed 2021-12-26 by haneefdm