Marus/cortex-debug #201: Can't launch GDB Server in latest vscode

5783mgordon, 2019-09-06:
I've been using cortex-debug for awhile and love it.  I've installed on a new system with vscode 1.38 and I can no longer debug at all.  I get the following error message:

"undefined GDB executable "/usr/bin/arm-none-eabi-gdb" was not found.  
Please configure "cortex-debug.armToolchainPath" correctly.

I added the "cortex-debug.armToolchainPath" to my settings.json file, I've double checked that arm-none-eabi-gdb runs correctly and is in fact in /usr/bin.

It also appears that the "Adapter Output" view is removed from the latest version of vscode so I can't find that output anymore to get insight into what is failing.

haneefdm, 2019-09-06:
The "Adapter Output" only appears if the extension could launch the server/gdb at least once. If basic executables can't be found, there is no attempt to even start them and hence not output.

I am assuming you set the `armToolchainPath` to the directory and not the path to gdb itself. You may want to attach your settings/launch json files.

As to why it is not working assuming you did everything right, not sure yet.

5783mgordon, 2019-09-06:
Thanks for the reply...
Here is the contents of my settings.json
```
{
    "cortex-debug.armToolchainPath": "/usr/bin",
    "cortex-debug.JLinkGDBServerPath": "/opt/SEGGER/JLink"
}
```
And here is the launch.json snippet
```
"version": "0.2.0",
        "configurations": [
            {
                "type": "cortex-debug",
                "request": "launch",
                "servertype": "openocd",
                "cwd": "${workspaceRoot}",
                "executable": "build/opfw/platforms/stm32/boards/gen300/gen300_nbl_Debug.elf",
                "name" : "OOCD Debug Gen300 ",
                "device": "STM32F777VI",
                "configFiles": [
                   "/usr/share/openocd/scripts/board/gen300.cfg"
                ],
                "gdbpath": "/usr/bin/arm-none-eabi-gdb",
                "svdFile": "../../STM32F7_svd_V1.4/STM32F7x9.svd",
                //"v1": false
            },
            {
                "type": "cortex-debug",
                "request": "launch",
                "servertype": "jlink",
                "cwd": "${workspaceRoot}",
                "executable": "build/opfw/platforms/stm32/boards/gen300/gen300_nbl_Debug.elf",
                "name" : "Debug Gen300",
                "device": "STM32F777VI",
                "interface": "swd",
                "svdFile": "../../STM32F7_svd_V1.4/STM32F7x9.svd",
                "ipAddress": null,
                "serialNumber": null,
                //"gdbpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
                "swoConfig": {
                    "source": "probe",
                    "enabled": true,
                    "swoFrequency": 4000000,
                }
            }
        ]
     }

I also downgraded VSCode to 1.33.1 (March 2019) and it's working fine with that version.  I have tried both the OpenOCD and JLink launch configurations and get errors with both.
Thanks.

haneefdm, 2019-09-06:
Hmm. I am using VSCode 1.37.1 and don't see an issue. That code that checks to see if the executable exists is not even VSCode API. It is simply Node.js

So, this happens regardless of if you set `gdbpath`? I never used `gdbpath` myself

5783mgordon, 2019-09-11:
I see the problem with our without gdbpath.  The newest version is 1.38.1.  I've backdated to 1.33.1 and I no longer have any issues.

rei-vilo, 2019-10-11:
Same issue here, with VSC 1.39.1 on macOS 10.15 _Catalina_.

[image]

The `gdb` has been duly authorised with `codesign`. It used to work fine with previous macOS 10.14 _Mojave_.

Are other users experiencing the same issue?

haneefdm, 2019-10-11:
That dialog says it can't launch OpenOCD the GDB server. Check the Output tab "Adapter Output" window to see messages from OpenOCD. Is the OpenOCD executable in your path? In your settings.json, you will need something like.
```
"cortex-debug.openocdPath": "/Applications/ModusToolbox_2.0/tools_2.0/openocd/bin/openocd"
```
You can also use `"serverpath"` in `launch.json` to specify the full path to openocd executable

I am using VSCode 1.39.1 but I am still on 10.14.6 (Mojave). Afraid to move to Catalina just yet. I doubt OS is the problem though.

haneefdm, 2019-10-11:
Please post output from OpenOCD from the Output Tab. If you want [to see where it is](https://github.com/haneefdm/cortex-debug-samples/blob/master/images/openocd-error.jpg)

rei-vilo, 2019-10-12:
@haneefdm Thank you so much for your answers.

Adding `"cortex-debug.openocdPath"` in `settings.json` did the trick. However, `"serverpath"` in `launch.json` was refused.

Case resolved!

closed 2020-03-29 by haneefdm