Marus/cortex-debug #202: Network connected BMP

sidprice, 2019-09-11:
I have a derivative of the Blackmagic Probe that is about to be launched as a Crowd Supply campaign. I use VSCode for my STM32 development and would love to be able to use this new probe. It is fully compatible with BMP, except, the GDB server is exposed over the network on the normal GDB port of 2159.

Is it possible to configure the extension to connect with this probe.

haneefdm, 2019-09-11:
Have you tried the `BMPGDBSerialPort` setting in `launch.json`. Technically it can be anything like `ip-addr:port#` or `host-name:port#` or `COM0` or `/dev/tty04` .. the myriad of remote connections that gdb supports.

If you can connect with command-line gdb, you can do the same with this extension.

haneefdm, 2019-09-13:
Not sure why you would see the difference between using COM vs TCP. You may want to set `runToMain` to false, set a bkpt. in main yourself and see what is going on. How many times do you have to press `continue` button before it hits main and what do you see each time it stops.

When `runToMain` is set to true, all the extension does is to set a temp. bkpt. at main issue a `continue` once ... but if there is an exception or some other reason your program halts before main, then the extension would not know. So, unless your programming method between COM and TCP is a bit different or has side effects you should see the same results.

sidprice, 2019-09-14:
After some careful examination of the logs it appears I have a bug somewhere in my code that is causing an exception. I will close this issue, many thanks @haneefdm for your help and patience.

closed 2019-09-14 by sidprice