Marus/cortex-debug #204: Enhance memory window...

forehead2k, 2019-09-13:
It would be a GREAT usability improvement if the memory window supported expressions (or at least a symbol) in addition to bare addresses. Having to round trip through a map file or nm output is... tedious.

haneefdm, 2019-09-13:
Normally, to get an address, I use the watch window and then copy the address value from there.

But between compilations, this constant address can change and one can easily miss that. GDB accepts an expression for the address but not for length/size, which should be good enough.

I looked at the [DAP protocol](https://microsoft.github.io/debug-adapter-protocol/specification#Requests_ReadMemory) and a `memoryReference` is a `string` which technically can be an expression. CortexDebug does not use this spec now for mem. reads, but wanted to be guided by this spec.

I can look at implementing this -- for the address only.

haneefdm, 2019-09-21:
I have a prototype working. Actually seems pretty useful and very little overhead or changes to the extension. If no objections, I can do a PR next week. Here is an example with a silly expression
[image]

haneefdm, 2019-09-21:
Probably not very useful (with threads or lower down the stack frames) but you can also view memory contents relative to a gdb recognized CPU register...

[image]

haneefdm, 2019-10-15:
Released in V 0.3.3

closed 2019-10-15 by haneefdm