Marus/cortex-debug #205: Feature requests

howroyd, 2019-09-17:
Couple of feature request for this excellent extension, please:

1.  In preRestartCommands (or similar) a command to stop the current debug session, if running.
2.  A "continue" option, as an alternative to "runToMain" so that when a debug session starts it just goes straight into it without pausing.

...coming from Eclipse and quite used to the second feature at least.

Cheers

haneefdm, 2019-09-17:
About Feature 2, I just implemented a way to do this but more general-purpose than Eclipse. See Issue #197. So, it will be there in the next release or two.

About Feature 1, could you explain this a bit more? And, stop as in terminate the session?

howroyd, 2019-09-18:
Awesome, ty very much.
Yeah so if already in a debug session, clicking restart (ctrl+shift+f5) triggers a re-flash of the chip.  What I am after is just a processor restart without reflashing, if that makes sense?

haneefdm, 2019-09-18:
Not sure why you are thinking a `restart` would `re-flash`.

Unless you are using custom overrides or adding commands to re-flash using pre/post restart, the default behavior is to **not** reflash the device for a restart. I also not recommend a re-flash because that is the current purpose of restart.

So, my answer is that a restart does not re-flash the device or even re-load the executable/symbols into gdb in case the executable changed. That is what the Stop + Start will do (two separate actions/commands). For a restart, by default, the only thing that is done is to `reset halt` the device.

Could you post your launch.json and/or some evidence of re-flashing? I will take a look.

howroyd, 2019-09-18:
Ah I must be mistaken then, apologies.

For reference, my launch.json for coretex-debug:
`{
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "executable": "./build_hub_dev/hub_dev.elf",
            "name": "Debug (OpenOCD)",
            "device": "STM32F413VH",
            "configFiles": [
                "C:\\Users\\howro\\AppData\\Roaming\\xPacks\\@gnu-mcu-eclipse\\openocd\\0.10.0-12.1\\.content\\scripts\\interface\\stlink-v2.cfg",
                "C:\\Users\\howro\\AppData\\Roaming\\xPacks\\@gnu-mcu-eclipse\\openocd\\0.10.0-12.1\\.content\\scripts\\board\\stm32f4discovery.cfg"
            ],
            "runToMain": true,
            //"rtos": "FreeRTOS",
            "preRestartCommands": [
                "file ./build_hub_dev/hub_dev.elf",
                "load",
                "enable breakpoint",
                "monitor reset"
              ]
        }`

haneefdm, 2019-09-18:
Yup, found your issue. Your first line `file ...` is reloading the elf file into gdb and the next line `load` is re-flashing the device.

```javascript
            "preRestartCommands": [
                "file ./build_hub_dev/hub_dev.elf",
                "load",
                "enable breakpoint",
                "monitor reset"
            ]
```

I guess the last two lines is to continue and stop in main? Perpahs this should be done in postRestartCommands? This extension already does a 'monitor halt' nullifying your reset??

closed 2019-09-18 by howroyd

haneefdm, 2019-09-18:
Thanks, this is all @Marus stuff. Could I request that you close this issue?