Marus/cortex-debug #207: Issue handling paths with non-ascii characters on Windows (?)

vilvo, 2019-09-20:
Extension works great with GDB on my setup. 
I encouraged a colleagues to use it as well. 
Setups are identical except for a non-ascii character in his Windows 10 UserAccount and starting the debugger results in:

"c:/Users/AColleagueWith_An_ä_In_His_Name/.vscode/extensions/marus25.cortex-debug-0.3.1/support/gdbsupport.init: No such file or directory."

haneefdm, 2019-09-20:
I think this a problem with gdb itself. I think you will run into other problems over time with non-ascii chars in path names. Does your basic debug session still work? If so, there can be a workaround.

vilvo, 2019-09-20:
Thanks for the quick comment!

On my setup GDB works from the path
`"serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe"`
under `launch.json` definition - similarly from the Windows CMD. 

Given that, I could it be that sourcing the gdbsupport.init on this line 
https://github.com/Marus/cortex-debug/blob/afab539f1a8504647ed5b233c2bd89d9756ceb86/src/gdb.ts#L308
is not platform independent on handling the path - not escaping non-ascii?

I agree that non-ascii in paths is calling for trouble, but I'm just looking to help a colleague who happens to have a non-ascii in his username.

haneefdm, 2019-09-20:
I am not even sure how to escape Unicode chars for gdb machine interface which is plain text. It is not like HTML or C strings where you can do \uXXXX. The code you pointed out uses `extensionPath` which is supplied by VSCode (we change backslashes to forward slashes). Other tools and even Linux has the same issues.

https://github.com/Microsoft/vscode-cpptools/issues/1998
https://github.com/Microsoft/vscode-cpptools/issues/602

See if you can manually start gdb command line and find a way to source that file.

The only way I know how to handle Unicode paths on Windows for tools like gdb is to convert all pathnames to 8.3 DOS format but that is not a good solution.

vilvo, 2019-09-20:
Thanks,

I created a symlink with non-ascii to start gdb as follows:
`C:\>mklink /D C:\äää C:\Users\MyUserName\.vscode\extensions\marus25.cortex-debug-0.3.1\support\
symbolic link created for C:\äää <<===>> C:\Users\MyUserName\.vscode\extensions\marus25.cortex-debug-0.3.1\support\`

Tested with:
```
C:\Program Files (x86)\GNU Tools ARM Embedded\7 2017-q4-major\bin>arm-none-eabi-gdb.exe --init-command="C:\äää\gdbsupport.init"
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb)
```

vilvo, 2019-09-20:
But you are right, sourcing the same file does not work as gdb converts `ä`s to ascii.

```
C:\Program Files (x86)\GNU Tools ARM Embedded\7 2017-q4-major\bin>arm-none-eabi-gdb.exe
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) source
source command requires file name of file to source.
(gdb) source C:\"""\gdbsupport.init
C:\"""\gdbsupport.init: No such file or directory.
(gdb)
```

vilvo, 2019-09-20:
If cortex extension would support configuring the path of `gdbsupport.init` instead of assuming it from the extension path (always under UserAccount/.vscode/) - the symlink could be used to remove the non-ascii from the path.

haneefdm, 2019-09-20:
Btw, with gdb, always use forward slashes.

That file is only the beginning of issues. You will see many more. Also, the extension does not decide where the info lives. VSCode does and it affects every extension.

VSCode does offer a command-line option to specify extension-dir... You can launch VSCode like.
```
code ---extensions-dir <dir>
```
My suggestion is to change the User's folder-name. It is bit complicated but in the long run, it is best. You can avoid the "death by a thousand cuts" problem :-) Anything compiled with mingw or cygwin or even Windows programs are known to fail.

haneefdm, 2019-09-20:
Help from `code -h`
```
  --extensions-dir <dir>                            Set the root path for extensions.
```

vilvo, 2019-09-20:
Thanks @haneefdm

closed 2019-09-20 by vilvo