Marus/cortex-debug #208: Atmel SVD files not working.

martin31821, 2019-09-24:
I tried to load the newest SVD file for the ATSAM C21 and I'm seeing the following error message:
```
mainThreadExtensionService.ts:65 TypeError: Cannot read property '0' of undefined
	at Function.parsePeripheral (/home/martin/.vscode…tension.bundle.js:1)
	at module.exports.parseSVD.u.readFile.c.parseString (/home/martin/.vscode…tension.bundle.js:1)
	at h.<anonymous> (/home/martin/.vscode…nsion.bundle.js:211)
	at h.emit (events.js:182)
	at r.onclosetag (/home/martin/.vscode…nsion.bundle.js:211)
	at _ (/home/martin/.vscode…nsion.bundle.js:212)
	at F (/home/martin/.vscode…nsion.bundle.js:212)
	at B (/home/martin/.vscode…nsion.bundle.js:212)
	at r.write (/home/martin/.vscode…nsion.bundle.js:211)
	at h.module.exports.t.Parser.h.parseString (/home/martin/.vscode…nsion.bundle.js:211)
	at h.module.exports.p [as parseString] (/home/martin/.vscode…nsion.bundle.js:211)
	at Object.module.exports.t.parseString (/home/martin/.vscode…nsion.bundle.js:211)
	at module.exports.parseSVD.u.readFile (/home/martin/.vscode…tension.bundle.js:1)
	at FSReqWrap.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:53)
```

haneefdm, 2019-09-24:
Could you attach a link to the svd file?

martin31821, 2019-09-24:
http://packs.download.atmel.com/Atmel.SAMC21_DFP.1.2.176.atpack 
(ATPack is a tar or zip, I could just extract it)

haneefdm, 2019-09-24:
It is a zip file with a whole bunch of SVD files. Which particular one were you using? The problem probably exists with most/all of them, but it would make it faster for me to know.

martin31821, 2019-09-24:
yeah, sorry. I'm using ATSAMC21E18A.svd

haneefdm, 2019-09-24:
It is a problem with a peripheral "PTC" with no registers at all. I fixed the problem and hopefully will be available in the next release of Cortex-Debug.

A workaround is to delete the peripheral "PTC" or add an empty registers block. "PTC" has nothing else interesting for this extension.

martin31821, 2019-09-24:
Thanks for the quick fix, I just tried the branch and it works fine!

closed 2020-01-28 by haneefdm