Marus/cortex-debug #210: SVD not updating registers

mes1234, 2019-09-29:
Hello,

I've started using cortex debug addin.
It works ok, until I want to use svd file.
It looks like it not updating registers, after refresh it shows:
GDB -> App: {"token":267,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Unable to read memory."]]}}

Here is my code repo:

https://github.com/mes1234/nucleo_bare_metal

haneefdm, 2019-09-29:
Could you post a little bit more from the Debug Console window?

mes1234, 2019-09-29:
Sure, I'll do it when I get to laptop

mes1234, 2019-09-29:
Some details:
I am working on ST nucleo board with STM32F103RBT6 unit.
At this stage I am working on bare metal approach to blink LED on GPIOA pin 5. Code is running ok and LED is blinking.
I've tried to add svd file for my unit. It starts and shows all registers correctly mapped to memory, according to spec GPIOA is 0x4001 0800 and it shows this way in svd preview in VS code. 
Also I am able to run full debugger in VS code using preview of registers which updated according to program.
The issue is with svd preview. It does not update value after any operation. eg. 
```c
GPIOA->ODR =0x000ffff;
```
Works correctly in code but register is still:
[image]

Also after any operation of refreshing svd preview in debug console I receive:
[image]

This is my repository:
https://github.com/mes1234/nucleo_bare_metal
 and this is my .vscode/launch.json:
```js
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "pyocd",
            "cwd": "${workspaceRoot}",
            "executable": "exec/main.elf",
            "name": "Debug (pyOCD)",
            "device": "STM32F103RB",
            "targetId": "",
            "boardId": "",
            "showDevDebugOutput": true,
            "svdFile": "svd/STM32F103.svd",
            "preLaunchTask": "build",
        }
    ]
}
```

Is there any other setup I need to complete to allow svd preview to update?
I am running Ubutu 18.04 and pyocd as server.

Thank in advace for any help!

Witek

haneefdm, 2019-09-29:
Since I don't have your HW, I can't debug with your project but I can see what tools you are using.

What I need is the full output of the Debug Console... before the following. I want to know what requests were made to gdb, which ones succeeded and which failed. What I see below is just the response to one failed request.
```javascript
GDB -> App: {"token":267,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Unable to read memory."]]}}
```
And, the output from pyocd.

You can also use the memory window to see what is at 0x40010800. If memory reads are not working in that region of the address space then, SVD can't work. I see a couple of <Write Only> registers there that night be the issue. I still need to see what requests were made to GDB

mes1234, 2019-09-29:
Hi 
Here is whole output.

I cannot see memmory window. This is how my VS code looks like where I can find it?
[image]

```c
Please check OUTPUT tab (Adapter Output) for log of pyocd-gdbserver
Launching server: "pyocd-gdbserver" "--persist" "--port" "50000" "--reset-break"
1-gdb-set target-async on
2-interpreter-exec console "source /home/witek/.vscode/extensions/marus25.cortex-debug-0.3.1/support/gdbsupport.init"
3-target-select extended-remote localhost:50000
4-interpreter-exec console "monitor reset halt"
5-target-download
6-interpreter-exec console "monitor reset halt"
7-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from /home/witek/CODE/arm/test_gnu/exec/main.elf...\n"}]}
Reading symbols from /home/witek/CODE/arm/test_gnu/exec/main.elf...
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"0x08000162 in main () at src/main.c:22\n"}]}
0x08000162 in main () at src/main.c:22
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"22\t    if (val>32000)\n"}]}
22	    if (val>32000)
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["frame",[["addr","0x08000162"],["func","main"],["args",[]]]],["file","src/main.c"],["fullname","/home/witek/CODE/arm/test_gnu/src/main.c"],["line","22"],["arch","armv7"]]}]}
Not implemented stop reason (assuming exception): undefined
GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"connected","results":[]}}
8-thread-list-ids
GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":5,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":5,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
GDB -> App: {"token":5,"outOfBandRecord":[{"isStream":false,"type":"status","asyncClass":"download","output":[]}]}
9-thread-list-ids
GDB -> App: {"token":5,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["address","0x8000186"],["load-size","460"],["transfer-rate","32280"],["write-rate","230"]]}}
GDB -> App: {"token":6,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":7,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
10-break-insert "/home/witek/CODE/arm/test_gnu/src/main.c:17"
11-break-insert "/home/witek/CODE/arm/test_gnu/src/main.c:25"
GDB -> App: {"token":8,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
12-thread-info 1
GDB -> App: {"token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
13-thread-info 1
GDB -> App: {"token":10,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","1"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x0800014a"],["func","main"],["file","src/main.c"],["fullname","/home/witek/CODE/arm/test_gnu/src/main.c"],["line","17"],["thread-groups",["i1"]],["times","0"],["original-location","/home/witek/CODE/arm/test_gnu/src/main.c:17"]]]]}}
GDB -> App: {"token":11,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","2"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x0800016e"],["func","main"],["file","src/main.c"],["fullname","/home/witek/CODE/arm/test_gnu/src/main.c"],["line","25"],["thread-groups",["i1"]],["times","0"],["original-location","/home/witek/CODE/arm/test_gnu/src/main.c:25"]]]]}}
GDB -> App: {"token":12,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 1"],["frame",[["level","0"],["addr","0x08000186"],["func","reset_handler"],["args",[]]]],["arch","armv7"]]]]]}}
GDB -> App: {"token":13,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 1"],["frame",[["level","0"],["addr","0x08000186"],["func","reset_handler"],["args",[]]]],["arch","armv7"]]]]]}}
14-thread-list-ids
GDB -> App: {"token":14,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
15-thread-info 1
GDB -> App: {"token":15,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 1"],["frame",[["level","0"],["addr","0x08000186"],["func","reset_handler"],["args",[]]]],["arch","armv7"]]]]]}}
16-data-list-register-names
[... cut ...]
0000623:INFO:gdbserver:GDB server started on port 50000
0000646:INFO:gdbserver:One client connected!
0000765:INFO:gdbserver:Attempting to load Argon
0000766:INFO:gdbserver:Attempting to load FreeRTOS
0000766:INFO:gdbserver:Attempting to load Zephyr
0000766:INFO:gdbserver:Attempting to load RTX5
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0000892:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 1024 bytes (1 page) at 9.20 kB/s

```

mes1234, 2019-09-29:
Ok I found memory check: this is output:
[image]

haneefdm, 2019-09-29:
Okay, I see the problem. The memory read is trying to read 28 bytes but 8 of those bytes are write-only.

You enter the following in the "Debug Console"
```
-data-read-memory-bytes 0x40010800 16
```
The above should succeed. If the above fail, then we have a different problem. Change 16 to 20 or 24 and it will probably fail.

Please confirm. It is a bug in Cortex-Debug and I can fix it after your confirmation.

mes1234, 2019-09-30:
Hi 
I've checked both 16 and 20 and it give the same result. But I am having problem to understand the output.
Could you take a look?
```c
-data-read-memory-bytes 0x40010800 16
```
[image]
```c
-data-read-memory-bytes 0x40010800 20
```
[image]

haneefdm, 2019-09-30:
Oh darn/puzzled. Can we even read 4 or 8 bytes? If not, nothing will work. At this point, I am suspecting problem may be pyocd? Is it possible for you to try with OpenOCD -- just as an experiment.

My theory was that up to 16 bytes from that address, everything should be fine. Are other peripheral addresses working or is it across the board failure?

Normally even if you have a ```write-only``` mem-mapped-regs, trying to read from the debug port should still work with undefined results but it depends on how the chip designer decided to implement.

mes1234, 2019-09-30:
Switching to openOCD works like charm :)
Thank you very much for help!

closed 2019-09-30 by mes1234

haneefdm, 2019-09-30:
Whew!!. You may want to file an issue with either pyocd or STM simply mentioning why memory reads fail and/or how they can expect SVD to even work. Use gdb as an example instead of this extension. I know it is a bit of work It will help a whole lot of people.

Chances are you will have the exact same issues with Eclipse-based IDEs

mes1234, 2019-09-30:
OK I will post issue on pyocd while I can easily link this issue.

kab4rt, 2020-07-26:
Although the issue is now closed I would like to add some feedback as I recently run into the same problem with STM32L5 and I think that the potential solution description may save some other people time.

Firstly I would like to confirm that Cortex-Debug plugin is working properly. Secondly the problem with SVD or more generally with peripherals memory sector access is caused by the way pyOCD is handling CMSIS packs, or to be more precise - by the  information provided in pack descriptor.
During GDB connection establishment some information are exchanged between client and server. One of it is memory map request (qXfer:memory-map:read) with response packet format as an XML document. Apparently this data is later on used by a GDB client to validate memory I/O operation before sending a request to a server. The received memory map can be dumped to client console by `info mem` command.
During my tests I was able to confirm that when peripheral address space is not present in the memory map a data read request packet is never transmitted to the GDB server. Peripheral register view works with openOCD because the memory map provided to the client provides r/w permissions to almost full MCU address area.
The last part that remains is to explain why memory map provided by pyOCD is incomplete. Basically, the number of targets supported by the tool after the installation is limited but can be easily expanded with CMSIS packs. Those packs are parsed during server initialization to determine flash sectors and programming algorithms and generally to build device memory map. Now let's take a look at some device defined in STM32L5 pack descriptor file (.pdsc):
```XML
<!-- *************************  Device 'STM32L552CCTx'  ***************************** -->
<device Dname="STM32L552CCTx">
    <debug svd="CMSIS/SVD/STM32L552.svd"/>
    <compile header="Drivers/CMSIS/Device/ST/STM32L5xx/Include/stm32l5xx.h" define="STM32L552xx"/>

    <book name="Documentation/DB3609_STM32L552.pdf"          title="STM32L552xx Data Sheet"/>

    <memory name="Flash-non-secure" access="rx"  id="IROM1" start="0x08000000" size="0x00040000" default="1" startup="1" />
    <memory name="Flash-secure"     access="rx"  id="IROM2" start="0x0C000000" size="0x00040000" default="0" startup="1" />
    <memory name="SRAM-non-secure"  access="rwx" id="IRAM1" start="0x20000000" size="0x00040000" default="1" init="0" />
    <memory name="SRAM-secure"      access="rwx" id="IRAM2" start="0x30000000" size="0x00040000" default="0" init="0" />

    <feature type="QFP" n="48"/>
</device>
```
Flash and RAM sectors are properly defined by that's it, all remaining address space is not part of memory map and therefore illegal to access by a GDB client. After I had modified the pack descriptor and added additional memory section for peripheral address space the SVD/peripheral register view functionality was fully restored.
```XML
<memory name="Peripheral"      access="rw" id="PERPH" start="0x40000000" size="0x10000000" default="0" init="0" />
```
The same issue type ("unable to read memory") will happen in case of other address sectors like FSMC external memory or some vendor registers unless they are not defined in the memory map. 

For the record my environment:
- GDB client: GNU Arm Embedded Toolchain 8.3.1
- GDB server: pyOCD 0.27.1
- STM32L5 pack: Keil.STM32L5xx_DFP.1.2.0.pack
- Cortex-Debug extension: 0.3.7

haneefdm, 2020-07-27:
Thank you for the investigation, resolution, and explanation.

AlexanderStein, 2020-08-19:
Another workaround when using pyOCD is to use a [user script](https://github.com/mbedmicro/pyOCD/blob/fa5992a542473ad5709e3f4d00baaca2ba724268/docs/user_scripts.md)
So you can add those regions manually without changing SVDs and packs