Marus/cortex-debug #214: Feature request: Change core registers to be displayed hexdecimal

sergioagp, 2019-10-16:
Please show the rXX in hexadecimal or at least give the option to change between formats.
I stopped using cortex-debug because of this issue. I would like to use it again.

haneefdm, 2019-10-16:
For now, could you try the watch window. You can do `$r3,x` and it will display `r3` in hex. The register window uses the natural format (as specified by ARM/gdb) which does a better job, especially when dealing with floating point registers.

marduino, 2019-12-24:
@haneefdm 
Sorry for jump in.

I have the same issue and I would agree with sergioagp, in most cases it makes no sense to display non-hex format for registers, as they normally read together with dis-assembly code.

Would be good at least provide an option to allow select hex.

Thanks!

haneefdm, 2019-12-24:
@marduino No problem. Agree about an option.

The format used is the (natural) format recommended by ARM. Things like sp, pc, lr are hex, lower rXX are supposed to be general-purpose integer variables, etc. When it comes to floating point registers (sXX, dXX)  though, it becomes super important. You don't want to see them in hex. Granted most Cortex users are not interested but for those who do, it is nearly impossible to convert on the go.

I will do what I can to help implement with permission from @Marus ANd, there are limits for what we can do with the VSCode API and still being performant

marduino, 2019-12-25:
Thanks @haneefdm , I just saw the another thread about this, 
FYI. https://github.com/microsoft/vscode/issues/28025
I tried the command "set output-radix 16" from gdb terminal, it helps a lot.

Aerijo, 2020-02-13:
Version 0.2.7 allows individual configuration on right clicking the register, so it seems to have been removed at some point. Shouldn't it be possible to support this, and add an additional option "as returned by GDB" to the Auto / Hex / Decimal / Binary list? Or was this a performance problem?

haneefdm, 2020-02-13:
In the next release, you can have an option to display everything in hex or natural. Details in ChangeLog.

Per register config? Maybe. Stay tuned. A bit of work to get it right.

biovoid, 2020-02-15:
It's very important to be able to setup default view of core registers and variables. Also current UI of registers view requires a bit visual alignment if possible.

closed 2020-03-30 by haneefdm