Marus/cortex-debug #215: Jlink debug session silently fails

Dirbaio, 2019-10-18:
When I click the "Start Debugging" button, the progress indicator runs for ~1 second, then stops, as if the debug session had finished. I can't see any indication on why it fails.

After this, the JLinkGDBServer is no longer running (checked with `ps aux`).

Interestingly in the output JLinkGDBServer doesn't get to print `Waiting for GDB connection...`, but if I run the exact same command in a terminal, it does. maybe if it's crashing or getting killed?

Any input on how can I debug this would be appreciated!
Thanks :)

---

Code 1.39.2
cortex-debug 0.3.3

settings.json has this (I have `arm-none-eabi-gdb` and `JLinkGDBServer` in my PATH though. I've tried with and without these settings and the result is the same)
```
    "cortex-debug.armToolchainPath": "/opt/gcc-arm-none-eabi-8-2018-q4-major/bin",
    "cortex-debug.armToolchainPrefix": "arm-none-eabi",
    "cortex-debug.JLinkGDBServerPath": "/usr/bin/JLinkGDBServer"
```

launch.json:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/build/controller.out",
            "name": "Debug (J-Link)",
            "device": "NRF52840_XXAA",
            "interface": "swd",
            "ipAddress": null,
            "serialNumber": null,
            "showDevDebugOutput": true,
            "svdFile": "${workspaceFolder}/nrf_sdk/modules/nrfx/mdk/nrf52840.svd"
        }
    ]
}
```

Debug console output:
```
Please check OUTPUT tab (Adapter Output) for output from /usr/bin/JLinkGDBServer
Launching server: "/usr/bin/JLinkGDBServer" "-if" "swd" "-port" "50003" "-swoport" "50004" "-telnetport" "50005" "-device" "NRF52840_XXAA"
Launching GDB: "/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
1-gdb-set target-async on
```

Adapter Output:
```
SEGGER J-Link GDB Server V6.52d Command Line Version
JLinkARM.dll V6.52d (DLL compiled Oct 15 2019 16:16:51)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52840_XXAA
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 NRF52840_XXAA
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jan  7 2019 14:07:15
Hardware: V1.00
S/N: 683457147
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 50000
```

haneefdm, 2019-10-19:
Hmmm, as you suspect, it may be crashing. Could you replace/intercept server executable with a shell script and see if it is crashing and run the actual server from the script to investigate further?

Dirbaio, 2019-10-21:
Thanks for the suggestion. I've tried with the below script (and variations of it):

```sh
#!/bin/bash

echo Args: $@

trap "echo GOT SIGINT!" SIGINT
trap "echo GOT SIGTERM!" SIGTERM

JLinkGDBServer $@

echo Exited with code $?
```

The result is the same. 

I do see the "Args: " line printed, but not the `Exited with code` nor the `GOT SIGINT/SIGTERM` ones. I don't think JLinkGDBServer is crashing, it seems something is killing the process with SIGKILL (not SIGINT/SIGTERM).

I've also tried with OpenOCD, and the same issue happens (OpenOCD starts, does things, then gets killed).

Dirbaio, 2019-10-21:
I've done more digging running the extension from source.

If I debug the debug server (with `"debugServer": 4711` in launch.json) I get this in the debug console:

```
Please check OUTPUT tab (Adapter Output) for output from JLinkGDBServer
Launching server: "JLinkGDBServer" "-if" "swd" "-port" "50003" "-swoport" "50004" "-telnetport" "50005" "-device" "NRF52840_XXAA"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
1-gdb-set target-async on
2-interpreter-exec console "source /home/dirbaio/cortex-debug/support/gdbsupport.init"
3-target-select extended-remote localhost:50003
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
8-enable-pretty-printing
undefinedarm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

If not running with a debugger, output stops at `1-gdb-set target-async on` like before. 

It seems that on exit the log output is not fully flushed before exiting for some reason. Either way, now I do see the root cause error, I'm going to try fixing it.

Dirbaio, 2019-10-21:
Installing `ncurses5-compat-libs` from the AUR fixed it!

For the record: I'm using the prebuilt toolchain from ARM: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads which apparently GDB requires an old ABI of ncurses.

Should I close this? If so maybe we should open a "logs can be swallowed on debug server exit" issue

haneefdm, 2019-10-21:
Thanks for chasing it down. I will see what we can do about the logs...

closed 2020-03-29 by haneefdm