Marus/cortex-debug #216: "Error: couldn't bind tcl to socket on port 6666: Address already in use"...

busticated, 2019-10-21:
hey there :wave:

i'm running into a new but familiar error using the recently released `v0.3.3` update. here are the steps to reproduce:

1. start a new debugging session
2. upon completion, click "disconnect"
3. start another debugging session

**expected**
debugging session starts, i'm able to pause and step around

**actual**
debugging session fails to start, i get a error dialog saying "Failed to launch OpenOCD GDB Server: Timeout"

having run into this from time to time in the past, i know to check the "Adapter Output" within VSCode where i see:

```
Open On-Chip Debugger 0.10.0+dev-00920-g6ea43726 (2019-09-09-19:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Error: couldn't bind tcl to socket on port 6666: Address already in use
```

running `lsof -i :6666` returns:

```
COMMAND    PID        USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
_openocd 22018 me    3u  IPv4  64035      0t0  TCP localhost:6666 (LISTEN)
me@ubuntu:~$ 
```

...which confirms that i have an instance of `openocd` running still despite having ended the debugging session (via the "disconnect" button in VSCode's debug toolbar).

if i quit that process (via `kill <pid>`), i'm able to start a new debugging session in VSCode without error.

i've seen this in the past (https://github.com/Marus/cortex-debug/issues/81#issuecomment-449185068 and https://github.com/Marus/cortex-debug/issues/181) though the later case was due to my copy of `openocd` being weird. whereas in the past this would happen something like once every 10 tries (running `cortex-debug@0.3.1), it seems to happen 100% of the time now (running `cortex-debug@0.3.3`) 

i've confirmed this behavior on:
* macOS `v10.13.6`
* ubuntu: `v18.04.3`

busticated, 2019-10-21:
just tried again over on macOS `v10.13.6` and i can confirm `cortex-debug@0.3.1` performs as expected while `0.3.3` leaves the `openocd` process running (and thus cannot reconnect to port `6666`)

haneefdm, 2019-10-22:
Hi @busticated is your debug session type a `"launch"` or an `"attach"`

Also, what version of OpenOCD are you running? Could I have a copy of it (the whole directory). There are many (private) versions of it floating around. It will help me debug.

busticated, 2019-10-22:
thanks for taking a look @haneefdm :pray:

> is your debug session type a "launch" or an "attach"

my debug session type is `"attach"`

> what version of OpenOCD are you running?

per https://github.com/Marus/cortex-debug/issues/166 we rebuilt our `openocd` and it now reports: `0.10.0+dev-00920-g6ea43726 (2019-09-09-19:42)` as it's version

happy to work with you on this however you prefer - i think the easiest option is to install the Particle Workbench extension i maintain:

https://marketplace.visualstudio.com/items?itemName=particle.particle-vscode-pack

once you have that, you can do the following:

1. if prompted to "install dependencies", click the "install" button and confirm the toolchain is successfully installed
2. run the `Particle: Create Project` command and follow the prompts
3. wait for the project to be loaded

after that, you'll find our customized `launch.json` within your new project's `./.vscode` directory and our `openocd` executable in your `~/.particle/toolchains/openocd/` directory.

haneefdm, 2019-10-22:
I tried and failed with the board I have. My board uses an interface that is not supported by Particle's version of openocd. So, I used a very old version of openocd and detach (disconnect) works as expected. I have to look to see how else I can replicate/fix the issue.

busticated, 2019-10-22:
sent you an email @haneefdm

haneefdm, 2019-10-26:
I have the boards @busticated. Thanks. Which board should I get started with?

busticated, 2019-10-26:
@haneefdm ah, nice! argon is probably the most straight-forward - here are some docs:

argon: https://docs.particle.io/argon/ ([setup walk-thru](https://docs.particle.io/quickstart/argon/#1-set-up-your-argon))
workbench: https://docs.particle.io/workbench/ ([debugging argon](https://docs.particle.io/tutorials/developer-tools/workbench/#debugging-3rd-generation-))

haneefdm, 2019-10-27:
I was able to duplicate the issue. It happens when the disconnect button is pressed while the program is running. If you pause and disconnect it was okay. Nevertheless, it needs to be fixed.

haneefdm, 2019-10-30:
@superjax Error msg. may be the same but reason could be very different. What are your steps to reproduce? What @busticated reported was `attach -> .... -> continue -> disconnect`. I could not reproduce it any other way. The first step and the last two steps were important.

haneefdm, 2019-10-30:
@busticated I pushed (V0.3.4) out a new release that should fix the issue you saw. When you get a chance, please confirm and close this issue. Thanks for reporting the problem and helping me debug.

busticated, 2019-10-31:
confirmed, `v0.3.4` is working much better now - thanks so much @haneefdm

closed 2019-10-31 by busticated

superjax, 2019-10-31:
`v0.3.4` fixed my issue as well.  Thanks @haneefdm!

doozMen, 2019-12-12:
I also had the issue but would like to add that v0.3.3 or any older versions did not fix everything for me as my device was previously configured for mesh. I found a solution after multiple resets. Elaborate info for others can be found [particle forum](https://community.particle.io/t/issue-particle-flash-for-debug-fails-and-puts-device-in-error-state-also-related-to-failed-to-launch-openocd-gdb-server-timeout/53705/3)

But what I did was

1. [install backup firmware](https://docs.particle.io/tutorials/device-os/led/argon/#firmware-reset) via dfu (reset halted otherwise) 
2. [factory reset](https://docs.particle.io/tutorials/device-os/led/argon/#factory-reset)
3. manually put in defu mode with buttons, particle debug command halts otherwise
4. start degugging session.

this works now. thanks @busticated for all the help