Marus/cortex-debug #221: SVD Register value got flipped with adjacent register.

arive298, 2019-11-21:
I created an SVD file for the RX130 micro and I noticed that for some registers the values were flipped. For example the `CMCNT` register at address offset `0x2` of the cmt0 peripheral at `0x88002` was reading the value that was supposed to be for the `CMCOR` register at offset `0x4`. Interesting enough the first register `CMCR` had the correct value.
[image]

I read out the entire peripheral at its base address with gdb and my suspicion was correct. Somehow these two register values got flipped.
[image]

I tried writing to each one and the value I wrote to one register was saved to the other register. 
[image]

Also for other registers, the values did not match what was in memory either.
[image]

 
I tried to debug the extension using the VS Code extension host(Used Launch Extension Target, Debug Server Target was throwing a weird error) to find out what was going on, but I encountered too many issues I could not resolve. Is there any way you could help me with this?

haneefdm, 2019-11-21:
You have registers at offset of 0x2? Very unusual. Usually, registers are at multiples of 0x4 offsets. But it should still work. We may have a bad assumption somewhere. Could you post a link to your svd file? (or attach it).  I will do what I can to get a quick resolution.

Also, I use the following target for debugging `Extension + Debug Server`

arive298, 2019-11-21:
Yes I know it's weird but that's what was defined in the datasheet. This is the SVD File
[rx130.svd.zip](https://github.com/Marus/cortex-debug/files/3875419/rx130.svd.zip)

haneefdm, 2019-11-21:
Thanks, I will take a look in a couple of hours. Looks like we have registers or all kinds of sizes.

arive298, 2019-11-21:
Yeah, I know its kind of crazy. I tried running the `Extension + Debug Server` and I get this error even though the `Extension Development Host` still pops up, and that file exists in that path. 
[image]

I then open my project and try to run the debugger and I get this error.
[image]
I don't know if the new VS Code update might have messed this.

haneefdm, 2019-11-21:
Try re-compiling everything. Using Command Pallette, `Tasks: Run Task`, and then `npm: test_compile`. And then run the task `npm: watch`. I am assuming you have already done an `npm install` in the root of the cloned repo directory to bring in all the external dependencies.

arive298, 2019-11-21:
Thanks. I already figured it was a build issue. I just did not know how to build it.

haneefdm, 2019-11-21:
This is the rare time I am seeing a big-endian processor/svd file :-) Sure, lots of cores endianness is configurable but this choice seems interesting.

But, I am not sure that is what is causing this problem. I see there is some evidence (not proof) that code base may be assuming little-endian as I look at the parse routines after reading the memory. Regardless of endianness, you should not see what you are seeing. __Sure bytes may be in the wrong order but not entirely wrong values?!?!__

Since I don't have a big-endian processor, not sure how far I can go. But I will try to make sure it is not some inherent issue.

```xml
  <cpu>
    ...
    <endian>big</endian>
    ...
  </cpu>
```

Marus, 2019-11-21:
This is certainly an interesting case; while in theory Cortex-M devices can be big or little endian - I'm unaware of any actual implementations that went the big endian route (although couldn't say 100% sure that there aren't any in existence). As a result of that I probably didn't pay much attention to endianness issues when I was coding the SVD parsing and processing (and certainly no testing of that as I don't know of an ARM Cortex-M device that I could have tested against).

Because of the fact that the field happens to be aligned on a 2 byte boundary instead of a 4 byte boundary - I could actually see values being swapped. As there could be some places where we are reading in values in bulk (instead of each individual register as a separate GDB command) and reading them 4 bytes at a time perhaps - although if that was the case I would have expected both the bytes within the registers to have been swapped as well as the register swap.

haneefdm, 2019-11-21:
Hi @Marus, welcome :-) Almost **all** of the SVD code I saw was pretty much endian-ness agnostic, except for one place where we convert bytes to values. But I should see a byte swap/order issue and not reading from the wrong offset which could mean that we have a deeper issue. I am digging into it.

It may take a bit of time as I have a busy day today.

Marus, 2019-11-21:
I might also try to take a bit of a look at this one - try to get back into the project a bit perhaps - will let you know if I find anything.

haneefdm, 2019-11-21:
Thanks @Marus 

This is what I found so far. I don't think we are decoding offset improperly. I created a simple svd file but mapped it to an SRAM address. I am seeing what I expect. In my FW, I did the following.

```C
volatile static uint32_t junk[] = {
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f,
    0x00010203, 0x04050607, 0x08090a0b, 0x0c0d0e0f, 
 } ;
```
I created a memory window that looks like this.
[image]

In my tiny svd file [xyz.txt](https://github.com/Marus/cortex-debug/files/3876479/xyz.txt), I plugged in addresses reported by the memory window

[image]

arive298, 2019-11-21:
So I have been searching around and I found out the issue is with GDB command `-data-read-memory-bytes`. When I run this command `-data-read-memory-bytes 0x88002 6` I get the wrong result: 
`{"token":120,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x00088002"],["offset","0x00000000"],["end","0x00088008"],["contents","c000bbbb9f8f"]]]]]}}`.

Using `x/6bx` command I read that same memory address and I get this:
[image]
which is the right answer. (Focus on 0xbbbb)
The `-data-read-memory-bytes` works fine if the length parameter is just 2 or 4, but it could give the wrong answer for other length parameter like 6 or 8. I think this issue might have to do with GDB reading memory from the RX130 micro so I'm going to try to find a workaround with the SVD file.
Thanks for the help

Marus, 2019-11-21:
ok - yeah - if `-data-read-memory-bytes` is reading things incorrectly on the platform all bets are off; and we have grouped things together specifically to try to avoid reading 2-4 bytes at a time.

That seems like a bug in the GDB implementation you are using for your platform. Almost wondering if the GDB is having an endianness issue (which is swapping full 4-by chunks around) - then our hardcoded little-endianness swaps the individual bytes for a 2-byte register around again (to get the individual register in the correct byte order).

Say if you had in 4 bytes of memory `0xDE 0xAD 0xBE 0xEF` - the first two is supposed to be one 16-bit register and the second two a second 16-bit register - if that was read as little-endian by GDB for some reason it would get into our stuff from GDB as `0xEF 0xBE 0xAD 0xDE`. Then we try to read them as two 16-bit little-endian values - so we get `0xBEEF` and `0xDEAD`. So you end up with the registers swapped - but each register value in the correct order.

Marus, 2019-11-21:
Are you sure you are overriding the settings from cortex-debug to launch the GDB provided by the toolchain for that Renesas chip? Or if there are special flags that maybe it needs to be provided to understand the correct endianness of the chip?

arive298, 2019-11-21:
Yeah GDB setup is fine. And actually the RX130 is little endian so my bad on that one. Anyways I re-tested with the little endian setup and still has the same issue.

Marus, 2019-11-21:
yeah- still seems like there is some issue in GDB and/or the GDB server for you setup though - as `-data-read-memory-bytes` sounds like it is giving the incorrect results. There is probably little we can do about that within the plugin unfortunately. 

Might be theoretically possible to work-around if we could determine exactly when this occurs and a proper function to do the mapping - but I don't like the idea of trying to put in hacks like this to work around what looks to be a legitimate bug in the GDB.

Changing the SVD to little endian wouldn't make any difference - as currently the plugin doesn't look at that - and assumes little in the places where we read a specific register value from the memory buffer (we probably should fix that for the hypothetical big-endian case).

arive298, 2019-11-21:
I agree this is not your issue, and I don't expect you to address it. Thank you so much for the help anyways.

Marus, 2019-11-21:
I'm going to close the issue for now. If you can get a GDB setup where `-data-read-memory-bytes` is working properly, and that doesn't fix the issue, then feel free to re-open it and we'll look into it more.

closed 2019-11-21 by Marus