Marus/cortex-debug #222: ITM logging may mistreat new-lines.

perlindgren, 2019-12-01:
Hi Folks
I'm using Cortex Debug for RTFM/Embedded Rust, and came across a problem with Cortex Debug decoding the ITM stream.
``` Rust
#![no_main]
#![no_std]

pub use firmware as _;

use cortex_m::{iprint, iprintln};

use panic_semihosting as _;
use rtfm::app;
use stm32f4xx_hal::stm32::Interrupt;

#[app(device = stm32f4xx_hal::stm32, peripherals = true )]
const APP: () = {
    struct Resources {
        itm: cortex_m::peripheral::ITM,
    }
    #[init]
    fn init(cx: init::Context) -> init::LateResources {
        let mut itm = cx.core.ITM;
        let stim0 = &mut itm.stim[0];
        iprint!(stim0, "123\n4");
        iprint!(stim0, "1\n2\n3\n4");

        rtfm::pend(Interrupt::EXTI0);
        init::LateResources { itm }
    }

    #[task(binds = EXTI0, resources = [itm])]
    fn exti0(cx: exti0::Context) {
        let stim0 = &mut cx.resources.itm.stim[0];
        iprintln!(stim0, "exti0 a long message no problem");
    }
};
```
The output gives:
```
[2019-12-01T19:54:31.622Z]   123
[2019-12-01T19:54:31.623Z]   41

[2019-12-01T19:54:31.623Z]   exti0 a long message no problem
```
It seems that the line break in the first `iprint!(stim0, "123\n4");` is handled correctly, but the second one `iprint!(stim0, "1\n2\n3\n4");` is somehow incorrectly handled. What's after the newline character is lost.

The problem is not so much when you directly output log data (followed by a newline), but I'm working on integrating a logging framework, where the data is streamed containing new lines in the middle of the stream sent, so I lost data (I thought). But, I believe the data is correctly sent, but incorrectly re-assembled by Cortex Debug.

Maybe the problem is already known...and there is some workaround available.

Thanks in advance.
  Per

perlindgren, 2019-12-01:
To clarify, if I run openocd and gdb from terminal I get the expected:
```
more /tmp/itm.fifo
123
41
2
3
4exti0 a long message no problem
```

haneefdm, 2019-12-05:
Sorry, I don't have an OpenOCD setup for ITM, but I tried to simulate the issue with a JLink. I see everything properly reported.

My C code below
```C
    SWO_PrintString("hello12345678901234567890\n");

    SWO_PrintString("123\n4");
    SWO_PrintString("1\n2\n3\n4");
    CyDelay(2000);
    SWO_PrintString("After a long delay\n");
```
The output seems right. I did it with and without that last line of C-code and I got expected results.
```console
[2019-12-05T18:34:14.625Z]   hello12345678901234567890
[2019-12-05T18:34:14.625Z]   123
[2019-12-05T18:34:14.625Z]   41
[2019-12-05T18:34:14.625Z]   2
[2019-12-05T18:34:14.625Z]   3
[2019-12-05T18:34:14.625Z]   4After a long delay
```

perlindgren, 2019-12-05:
Interesting, I don't know exactly what would differ in these scenarios. The Rust implementation to send data over itm, will try to fill whole itm packages (optimizing utilization). Maybe the C implementation sends one byte/character per itm package, not sure.

If that is the case, it may be that the decoder that gets confused.

haneefdm, 2019-12-06:
The C implementation definitely sends a packet at a time with each packet containing just one character and its header. Basically two bytes per char. sent.

haneefdm, 2019-12-06:
Could I ask you to try `od -a /tmp/itm.fifo` instead of `more` to look at the fifo file? That way, we can see both printable and other chars.

Marus, 2019-12-11:
Are you using rust’s ITM output here; rust does some grouping of multiple characters together (as technically each SWO message can be up to four bytes) that most frameworks do not do. Could be something specific to my decoding of those grouped chapters that is causing the mishandling of the new lines. I’ll try to take a look at that this weekend.

perlindgren, 2019-12-11:
Great. I've been playing around with a logging framework in Rust, and I first thought the problem was on the Rust side, but when running the ITM logging outside of vscode I did not see the problem, so I believe you are right on the spot. 

It would also be great if possible to have vscode viewing multiple ITM channels concurrently, (and remembering the layout of such views), not sure if that is even possible. For now there is a lot of clicking and drop down menus each time you start a new debug session, that is very annoying (but maybe not fixable as of today..)

The option to direct ITM data to external sinks (pipes perhaps) would also be great.

Marus, 2019-12-11:
I am actually working on a number of ITM/SWO related things at the moment, one of which is redirection of the full stream, as well as individual ports to output files. 

As for layout, we’re a bit limited here unfortunately (at least in the short term) by VSCode, as their standard output channels are just in the one place - selectable by drop down.

haneefdm, 2019-12-12:
Nice @Marus

perlindgren, 2019-12-18:
The cli tools that we use you find at https://github.com/japaric/itm-tools. So you can test against these to see/verify that stream re-direction works.

Unless VSCode steps up when it comes to output views, using external tools for viewing SWO/ITM is the better option. I guess its on their TODO list, but maybe not at the highest priority. VSCode has about half the "market" for Rust development regarding IDE usage in general. Not sure how it looks in particular for embedded, but I can imagine that embedded developers are keen to use IDEs (many vendors provide Eclipse based IDEs for their products, so if you come from a C background you have likely been exposed to IDEs.) What I'm getting at is that good support in VSCode for embedded Rust could be a selling point for VSCode. You might use this as an argument when discussing with the VSCode team regarding output viewing (which is currently problematic in VSCode). If you want, I could also help in the argumentation with the VSCode team. Not sure how its organized. For embedded development we have teams, with bi-weekly meetings, to move stuff forward. How does it look in the case of VSCode?

perlindgren, 2020-01-08:
Just checking back to see if there is any progress on the issue.