Marus/cortex-debug #224: OpenOCD debug hangs indefinitely

UberLambda, 2019-12-17:
Hello! Cortex-Debug has stopped working with OpenOCD on my machine:
- The debug configuration hangs indefinitely (spinning bar on "debug" header)
 - `Please check OUTPUT tab (Adapter Output)` is present in the debug console, but no Adapter Output panel is spawned
- OpenOCD server does not seem to be launching at all anymore (but it used to in the same project; not sure what changed).

<details>
<summary>OpenOCD launch config that is not working anymore</summary>

```json
  {
            "name": "stlink",
            "type": "cortex-debug",
            "request": "launch",
            //"preLaunchTask": "build",
            "executable": "${workspaceFolder}/bin/Screen_firmware.ino.elf",
            "cpu": "cortex-m3",
            "device": "STM32F103C8",
            "servertype": "openocd",
            "serverpath": "/usr/bin/openocd",
            "configFiles": [
                "interface/stlink-v2.cfg",
                "target/stm3ir2f1x.cfg"
            ],
        }
```

</details>

<details>
<summary>Working (tested) Black Magic Probe config for the same project</summary>

```json
 {
            "name": "bmp",
            "type": "cortex-debug",
            "request": "launch",
            "preLaunchTask": "build",
            "executable": "${workspaceFolder}/bin/Screen_firmware.ino.elf",
            "cpu": "cortex-m3",
            "device": "STM32F103C8",
            "servertype": "bmp",
            "BMPGDBSerialPort": "/dev/ttyBMP",
        },
```

</details>

Launching OpenOCD and `arm-none-eabi-gdb` manually from a shell works as expected.

Cortex-Debug: v0.3.4  
OpenOCD: v0.10.0  
Platform: Manjaro Linux x64
(I already have tried installing `ncurses5-compat-libs` as from https://github.com/Marus/cortex-debug/issues/215 to no avail)

haneefdm, 2019-12-17:
Could you please paste the entire output of the Debug Console?

Did you try to restart VSCode? Super rare but it happened to me and eventually restarted VSCode.

UberLambda, 2019-12-18:
`Please check OUTPUT tab (Adapter Output) for output from /usr/bin/openocd`, sadly it's all there is there - Adapter Output is not in the output panel's dropdown. I tried restarting multiple times to no avail.

haneefdm, 2019-12-18:
In your launch.json, could you add a line `"showDevDebugOutput": true` and maybe we will see a little bit more!?!?

So, the 'Adapter Output' window never even starts. That gives me a bit of a hint. Thanks.

Even if the server does not start, VSCode should not be hanging. It should timeout after a few (20?) seconds.

Btw, with OpenOCD, I always had to use the `"searchDir"` setting in launch.json to specify the search path for the config files.

UberLambda, 2019-12-18:
@haneefdm `showDevDebugOutput` did not seem to do anything, but I now get the expected output with `searchDir` set! Not sure why it was working before without it though?

<details>
<summary>New tasks.json</summary>

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "bmp",
            "type": "cortex-debug",
            "request": "launch",
            //"preLaunchTask": "build",
            "executable": "${workspaceFolder}/bin/ScreenFirmware.ino.elf",
            "cpu": "cortex-m3",
            "device": "STM32F103C8",
            "servertype": "bmp",
            "BMPGDBSerialPort": "/dev/ttyBMP",
        },
        {
            "name": "stlink",
            "type": "cortex-debug",
            "request": "launch",
            //"preLaunchTask": "build",
            "executable": "${workspaceFolder}/bin/ScreenFirmware.ino.elf",
            "cpu": "cortex-m3",
            "device": "STM32F103C8",
            "servertype": "openocd",
            "serverpath": "/usr/bin/openocd",
            //"showDevDebugOutput": true,
            "searchDir": [
                "/usr/share/openocd/scripts",
            ],
            "configFiles": [
                "interface/stlink-v2.cfg",
                "target/stm32f1x.cfg"
            ],
        }
    ]
}

```

</summary>

I guess my issue is fixed but I would like to find the root cause :)

haneefdm, 2019-12-18:
Yes, will do. What was happening was what I suspected, so hopefully I can find the root cause. I think the server was dying/exiting before we could even check on it  :-) Thanks for your help.

I am not sure why it was working before. It never worked for me without the `searchDir` but I got some output though and the debugger did not hang. So, this is new to me. It could be a new VSCode behavior as well, that we have to figure out and workaround.

UberLambda, 2019-12-18:
Thanks for the help!

closed 2020-03-31 by haneefdm