Marus/cortex-debug #225: View Memory fails on address of structure

g8ecj, 2019-12-31:
Simple variables and arrays of simple types seem to be OK but even though a structure has a defined address, I can't view it (nothing happens after the Length prompt). Same goes for an array of structures.   e.g.   
    uint32_t Var1;     // Var1 is displayed OK
    uint8_t Array1[11];    //  Array1 is displayed OK
    typedef struct {
      uint32_t start;
      uint32_t end;
      uint16_t length;
    } list;
    list List1;     // List1 displays nothing
    list List2[11];    // List2 displays nothing
Trying the 'address-of operator' gets it very confused!

haneefdm, 2019-12-31:
Are you sure that these variables have not been optimized out by the compiler? We look at some very complex data structures as seen below
[image]

haneefdm, 2019-12-31:
I just saw the title of your report. It says `View Memory`. So this is not a `Watch` window issue. So, I tried `View Memory` as well and it works over there as well.

Note that when you say view memory, you have to enter a C expression that evaluates to an address.

For List1 ==> the expression would be `&List1`
For List2 ==> the expression would be `List2`

g8ecj, 2019-12-31:
Too much Christmas pudding I think!! It works as described above.

g8ecj, 2019-12-31:
Another brain cell filled with useful information ;)

closed 2019-12-31 by g8ecj