Marus/cortex-debug #226: Why does variable show "optimized out" but watch displays variable?

sslupsky, 2020-01-04:
I have noticed on several occasions that a local variable sometime displays "optimized out". If I add the variable to the watch list, then I can see its value.  In the screen shot below, see "cfg" in the variables and watch windows.

I am trying to understand why there is a difference.  I believe I have my optimizations turned off ("-Og").  

[image]

haneefdm, 2020-01-04:
I couldn't tell you if this is the case or not without an exact executable/source-code to test. But, I can tell you how the Variables and Watch windows could differ.

- The Locals section/scope of the Variables Window is **strictly** what locals/args are available for that function of a particular thread frame.
- The Watch window, however, is a bit more dynamic. It will match any variable starting from the current thread/frame to static or global.

So, if you have a variable of the same name as global as local and the local one of them is optimized out, you can see different results.

If you can share a small test case, that would be helpful.

sslupsky, 2020-01-05:
@haneefdm Thanks for the feedback.  In the screen shot above, the "cfg" variable is a local variable in a Zephyr device driver.  I have referenced the code I was debugging at the time I took the screenshot below taken from a DMA driver in the Zephyr project https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/dma/dma_sam0.c.  In this instance, I had a breakpoint inside the i2c_sam0_dma_read_start() and captured the screenshot above.  You can see that "cfg" is defined as a local variable.  So, I am unclear why it doesn't display in the Variables window.  The code was compiled with the "-Og" flag so I think the optimizations were turned off but perhaps something is still optimized.  But, why would it appear in the watch window and not in the variables window?  That particular variable (address 0x1c324) is correct.  I've include a snippet of the compiled code output below as well.  You can see that the pointer to "cfg" is put into r5 at d436 and then used at d438 and d446.  So, I do not think I understand what "optimized out" in this context means.

```
static bool i2c_sam0_dma_read_start(struct device *dev)
{
	struct i2c_sam0_dev_data *data = DEV_DATA(dev);
	const struct i2c_sam0_dev_config *const cfg = DEV_CFG(dev);
	SercomI2cm *i2c = cfg->regs;
	int retval;

	if (!data->dma) {
		return false;
	}

	if (cfg->dma_channel == 0xFF) {
		return false;
	}

	if (data->msg.size <= 2) {
		/*
		 * The last byte is always handled by the I2C ISR so
		 * just skip a two length read as well.
		 */
		return false;
	}

	struct dma_config dma_cfg = { 0 };
	struct dma_block_config dma_blk = { 0 };

	dma_cfg.channel_direction = PERIPHERAL_TO_MEMORY;
	dma_cfg.source_data_size = 1;
	dma_cfg.dest_data_size = 1;
	dma_cfg.callback_arg = dev;
	dma_cfg.dma_callback = i2c_sam0_dma_read_done;
	dma_cfg.block_count = 1;
	dma_cfg.head_block = &dma_blk;
	dma_cfg.dma_slot = cfg->read_dma_request;

	dma_blk.block_size = data->msg.size - 1;
	dma_blk.dest_address = (u32_t)data->msg.buffer;
	dma_blk.source_address = (u32_t)(&(i2c->DATA.reg));
	dma_blk.source_addr_adj = DMA_ADDR_ADJ_NO_CHANGE;

	retval = dma_config(data->dma, cfg->dma_channel, &dma_cfg);
	if (retval != 0) {
		LOG_ERR("Read DMA configure on %s failed: %d",
			DEV_NAME(dev), retval);
		return false;
	}

	retval = dma_start(data->dma, cfg->dma_channel);
	if (retval != 0) {
		LOG_ERR("Read DMA start on %s failed: %d",
			DEV_NAME(dev), retval);
		return false;
	}

	return true;
}
```
```
0000d420 <i2c_sam0_dma_read_start>:
{
    d420:	b5f0      	push	{r4, r5, r6, r7, lr}
    d422:	46de      	mov	lr, fp
    d424:	4657      	mov	r7, sl
    d426:	464e      	mov	r6, r9
    d428:	4645      	mov	r5, r8
    d42a:	b5e0      	push	{r5, r6, r7, lr}
    d42c:	b091      	sub	sp, #68	; 0x44
    d42e:	0006      	movs	r6, r0
	struct i2c_sam0_dev_data *data = DEV_DATA(dev);
    d430:	6883      	ldr	r3, [r0, #8]
    d432:	4698      	mov	r8, r3
	const struct i2c_sam0_dev_config *const cfg = DEV_CFG(dev);
    d434:	6803      	ldr	r3, [r0, #0]
    d436:	691d      	ldr	r5, [r3, #16]
	SercomI2cm *i2c = cfg->regs;
    d438:	682b      	ldr	r3, [r5, #0]
    d43a:	469b      	mov	fp, r3
	if (!data->dma) {
    d43c:	4643      	mov	r3, r8
    d43e:	6c5b      	ldr	r3, [r3, #68]	; 0x44
    d440:	4699      	mov	r9, r3
    d442:	2b00      	cmp	r3, #0
    d444:	d071      	beq.n	d52a <i2c_sam0_dma_read_start+0x10a>
	if (cfg->dma_channel == 0xFF) {
    d446:	7dab      	ldrb	r3, [r5, #22]
    d448:	2bff      	cmp	r3, #255	; 0xff
    d44a:	d070      	beq.n	d52e <i2c_sam0_dma_read_start+0x10e>
	if (data->msg.size <= 2) {
    d44c:	4643      	mov	r3, r8
    d44e:	6b5b      	ldr	r3, [r3, #52]	; 0x34
    d450:	469a      	mov	sl, r3
    d452:	2b02      	cmp	r3, #2
    d454:	d96d      	bls.n	d532 <i2c_sam0_dma_read_start+0x112>
```

haneefdm, 2020-01-06:
A couple of observations:
- AFAIK, -Og does not mean do not optimize. It means Optimize in a way that does not interfere (heavily) with the debugging experience. If you truly want no optimization, then -O0 -g is perhaps more appropriate, but in most situations, -Og is appropriate
- I see that `cfg` inside the function is not allocated on the stack. It went into `r3`? So, technically it got optimized out.
- But I don't know where the `cfg` in the watch window is coming from. It appears it is an alias of `i2c_sam0_dev_config_3`. But how/why?

Unless I have a full listing including the symbol table of that object file, I can't tell what is going on or what the compiler did or what this extension did.

haneefdm, 2020-01-06:
Could we try the following?
- Break at the same point where you are seeing this behavior. Once that happens, don't change the current frame (clicking in the `Stack` window can change the current frame).
- in the Debug Console, type `print cfg`. Also try `info symbol cfg` and report what you see
- You can also try other info commands like `info frame 0`, `info locals`, `info args`
We want to see what information is reported by gdb.

sslupsky, 2020-01-07:
Here is the output:
```
print cfg
$1 = (const struct i2c_sam0_dev_config * const) 0x1c614 <i2c_sam0_dev_config_3>

{"token":248,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```
```
info symbol cfg
i2c_sam0_dev_config_3 in section rodata
{"token":274,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```
```
info locals
data = 0x20001c14 <i2c_sam0_dev_data_3>

cfg = 0x1c614 <i2c_sam0_dev_config_3>

i2c = 0x42001400

retval = <optimized out>
dma_cfg = {dma_slot = 63
, channel_direction = 6
, complete_callback_en = 1
, error_callback_en = 0
, source_handshake = 1
, dest_handshake = 0
, channel_priority = 9
, source_chaining_en = 0
, dest_chaining_en = 0
, reserved = 0
, source_data_size = 0
, dest_data_size = 0
, source_burst_length = 7476
, dest_burst_length = 8192
, block_count = 4308992
, head_block = 0xe000ed00
, callback_arg = 0x20002dd0 <_kernel>
, dma_callback = 0x1041c000
}

dma_blk = {source_address = 0
, source_gather_interval = 2
, dest_address = 118
, dest_scatter_interval = 536878412
, dest_scatter_count = 11011
, source_gather_count = 1
, block_size = 536878412
, next_block = 0x12b15 <next_timeout+12>
, source_gather_en = 0
, dest_scatter_en = 0
, source_addr_adj = 3
, dest_addr_adj = 0
, source_reload_en = 1
, dest_reload_en = 0
, fifo_mode_control = 13
, flow_control_mode = 1
, reserved = 0
}

__func__ = <optimized out>
{"token":300,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```
```
info args
dev = 0x20007094 <__device_i2c_sam0_3>

{"token":326,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

haneefdm, 2020-01-07:
It appears that the local variable `cfg` got aliased to a global constant struct ptr `i2c_sam0_dev_config_3`. So did `data` but it is not a constant. But `info locals` shows `cfg` as being valid local and *not* optimized out.

Maybe I now have enough info to see if I can create this situation with my firmware. Unfortunately, I won't get much time until Tue next week as I am traveling.

One more thing, I wanted to make sure that you are still seeing different behavior in Variables vs. Watch Windows for `cfg`. I ask because I noticed that the values of `data` and `cfg` have changed from your initial report. This is fine but wanted to verify.

sslupsky, 2020-01-08:
That is correct.  The results are from two different debug sessions.  I think there were probably several compiles between too.

sslupsky, 2020-01-08:
Yes, the cfg pointer gets the address of a global struct  that contains some device driver info.

haneefdm, 2020-01-20:
Sorry, @sslupsky I tried for several hours over multiple days to duplicate this but of no avail. If you can narrow down the application to some generic C-code, I am sure we can figure out why and get a fix or it may not even be a bug. I don't have the setup you have.

What you are seeing is indeed strange. I have a hunch(es) but no way to validate it.

sslupsky, 2020-01-20:
Apologies @haneefdm I do not mean to cause a lot of work for you.  I'll give some thought to how I might come up with some generic code to replicate this.  I only started using the debugger when I started experimenting with the Zephyr RTOS.  So, I am not sure if I can replicate this outside of Zephyr.

haneefdm, 2020-03-07:
This should be (partially) addressed in the current beta release. One thing you will notice is that at the beginning of the function, gdb reports different stuff from if you take one single step. I don't know why and no way for me to address it. Cortex-Debug cannot do any more magic and it relies totally on what gdb reports.

closed 2020-03-07 by haneefdm