Marus/cortex-debug #228: st-link V3 support

EU1KY, 2020-01-09:
There is no way to debug latest STM32 Discovery boards in VS Code because ST-Link V3 is not supported by debugging extensions. Appropriate GDB server has appeared recently: EBLink https://github.com/EmBitz/EBlink . Could you please add EBLink GDB Server support?

MabezDev, 2020-01-13:
Openocd has had support for stlink v3 since december 2018: https://github.com/ntfreak/openocd/commit/0bad9a42a3b7a3160b264b6d05934c30d8482bf5

You may have to use openocd from git or something more recent than the last release.

Marus, 2020-02-09:
I would suggest that you try out the OpenOCD support as pointed out by @MabezDev 

I'm hesitant to try and get things work with the EBlink software you pointed out because 1) it is not available on all platforms we support (namely macOS, which is also my primary development environment) 2) it is not open source - meaning there is little possibility for getting it running there and 3) I currently do not have anything with an ST-Link V3 to test it against.

timblakely, 2020-03-11:
Just confirming that I've been using the V3 on a NUCLEO-G474RE just fine with it (after applying [this patch](http://openocd.zylin.com/#/c/4807/) that I expect to be merged shortly.

PhilippHaefele, 2021-09-12:
Hi,
It seems that this ticket is also fixed in the meantime. OpenOCD has support in for a while now and Cortex Debug also has support for ST-Link V3. The issue with the shared libraries seems to be handled in #354. So good to close this one 🙂

haneefdm, 2021-09-12:
Thanks, @PhilippHaefele

closed 2021-09-12 by haneefdm