Marus/cortex-debug #230: Any plan to adding support for RISC-V Processor

fanghuaqi, 2020-01-14:
Hi guys, do you have any plan to adding support for RISC-V processor debug support in VSCode, I think it could be quite similiar to cortex debug, since they are both supporting gdb and openocd.

Thanks
Huaqi

haneefdm, 2020-01-24:
Haha, I knew this will come up sooner or later. 99% of what Cortex-Debug does is so processor agnostic.

But we are not users of RISC-V and have no HW/SW to test with. No, I am not volunteering but it is one of those things that grows out of need/want.

Marus, 2020-02-07:
I'm not sure I agree with it being 99% - but a majority of functionality would be processor agnostic as @haneefdm said. The biggest area that wouldn't be processor agnostic would be ITM/SWO output (which you could just ignore that functionality, as by default that is all disabled unless you configure it in your `launch.json`).

That said i know some users have been using it with non-Cortex M platforms (using things like symlinks so it finds the right GDB) and writing their own SVD files for memory mapped peripherals. For example saw someone on Reddit using the extension with ESP-32 devices.

That said though - like Haneef I don't personally do any development on RISC-V platforms - so not something I personally am looking at taking up maintaining official support for - but I don't object to the idea of pull-requests that would add support for other tool-chains/platforms in a secondary manner (I have been doing a personal project with an ESP-32 - so am actually considering trying to get that working nicely with the plugin myself - although too many things and too little time to have tackled that yet).

haneefdm, 2020-02-07:
Okay, maybe not 99% but maybe 91.63%? :-) @Marus takes very little credit for his awesome job. And yeah, ITM/SWO are too specific anyway but not active by default. Heck, there are Cortex chips that don't have that functionality enabled.

**I think we should close this issue** because there is no way for us to debug/test anything. But if the RISC-V community can donate **real** resources, then welcome, right?

Even with a PR for RISC-V related things, I am wary about maintenance unless it is something generic. This thought is above my pay grade :-) Upto @Marus the boss. Gonna shutup.

Side Note:
Having used Cortex-Debug so much, I feel like we can even use it for native debugging (path names issues aside). It is better than VSCode's offering for C/C++ in soooo many ways. Yes, C++ needs a lot of improvement and I am working on it little at a time. Maybe I will create a prototype to show what I mean. But, I don't want to. I can't handle the support load.

Marus, 2020-02-07:
Yeah - I think the issue could be closed - and I'm willing to consider PRs to help with it - but would have to consider it carefully - as the focus is still intended to be for Cortex devices

closed 2020-02-07 by Marus