Marus/cortex-debug #232: No SVD File Loaded

dzarda, 2020-01-22:
This appears to be a recurring issue. Tried relative and absolute paths, tried forward slashes.

Using [SVD](https://stm32.agg.io/rs/stm32l0x3.svd.patched) from this project: https://github.com/stm32-rs/stm32-rs

haneefdm, 2020-01-22:
Maybe I can help. But a bit more context, please?

What recurring issue? What do you see? Assuming an issue with SVD, but what are you observing?

dzarda, 2020-01-22:
Please excuse me, updated the description. Not sure about reproducability.

haneefdm, 2020-01-23:
Hi @dzarda, I doubt this is a rust specific issue. Cortex-Debug doesn't even know what rust is. I could not figure out **exactly** which SVD file is an issue. Sometimes, this can happen if the SVD is un-parseable or there is a bug in the SVD parser. Please point to the SVD file if you can.

dzarda, 2020-01-23:
SVD link is in the description, also here: https://stm32.agg.io/rs/stm32l0x3.svd.patched

haneefdm, 2020-01-23:
Sorry, I did not notice the SVD link earlier. I can now duplicate the problem. There was a problem with the DMA1 peripheral (around line 756) and resulted in the SVD file parsing getting aborted.

Not sure why and investigating. Thanks.

haneefdm, 2020-01-24:
Okay, I found and fixed the issue. It had to do with a `derivedFrom` attribute on enum values. It was not supported/expected by Cortex-Debug. However, even the current spec is unclear as to how it **should** be interpreted.

https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_enumeratedValues

It appears that the references are global in scope and not referenced by its scope. ie, one enum-values can refer to totally another peripheral's register's field's, enum-values, and if simple names are used, the resolution is unclear. I have ** an implementation** that works according to how I interpret the spec., but others might disagree. Easily changed if I am corrected.

I will do a PR shortly. Don't think it has a chance of breaking anything.

dzarda, 2020-02-18:
Can this be merged @haneefdm ?

haneefdm, 2020-02-19:
Yes, I am closing this now. Thanks for the reminder.

closed 2020-02-19 by haneefdm