Marus/cortex-debug #236: Input in QEMU mode

ZZYSonny, 2020-01-26:
Currently in QEMU mode we can only see output from output panel. And there's no other way to give input to the program.

It will be nice if qemu-system-arm runs as a background tasks like how C++ does it. So we can add input through terminal panel.
[image]

ZZYSonny, 2020-01-26:
I am thinking of using extern mode and launch qemu myself. I can do that with gdb manually, but when I try to connect VSCode to it, I can't see it stopping at my breakpoint.

Marus, 2020-02-07:
Yes - this is something I had hoped to eventually be able to do. At the current time output windows (where the qemu output is channeled) are output only - no way to input there - and there is only a single debug console (which is needed for sending input to GDB for things that aren't exposed in the VS code interface).

There are now some ways to access terminals for extensions within the terminal tab - so I'm wondering if it might end up being possible for it to end up launching the QEMU process in a new terminal that we control - this would likely allow for input and output (with the extension not having to channel the output into one of the output tabs) - while allowing the automatic launching. Although this might have some challenges in dealing with different shells on different platforms.

As for your second problem - not sure why an externally launched QEMU launched session wouldn't work using the external type. If it's connecting with a proper GDB MI2 interface it should be able to set breakpoints.

ZZYSonny, 2020-02-07:
When I set "overrideLaunchCommands": [], external mode works fine. So now I am using tasks and external mode.