Marus/cortex-debug #237: C++ static variable in template class

senorclean, 2020-01-27:
This might be too specific but I just started using C++ with this plugin (which has worked great for C so far) and I'm having some problems with variables I want to watch. If I have a static variable as part of a class for instance, I can't just click on it and add it to my watch list (it will list it as unable to create). 

If I add the instance of the class (as long it's in scope) then I can see everything of course. But most of my classes are actually class templates and it seems like this plug-in (or maybe this is a GDB limitation) can't parse out the correct information. Here's a quick example:

```
accel_namespace {

template <class TTimer>
class Accelerometer {
  // class definition
  static uint8_t variable;
};

};
```

So regardless of where I am, "variable" will never be visible. I tried modifying the expression to "accel_namespace::Accelerometer&lt;Timer2&gt;::variable" and then I could read everything. Is there any way to not have to manually enter all of that each time I want to view a static member variable?

Also, if I use the accelerometer template class in a nested configuration for another class, is it possible to just add this instance to the watch list?

```
template <class TAccel>
class DoSomething {
  // class definition
  TAccel accel;
};
```

In this instance, I could just add "accel" and then see everything. I don't know if that's even possible though. Thanks!

haneefdm, 2020-01-27:
Maybe I am missing something. I see class (template) definitions but I don't see any variables declared with them.

Do you have an actual instance of this class and is a local, global or static variable?

senorclean, 2020-01-27:
How do you mean? In the first code snippet, I just have 

`static uint8_t variable`

as an example. In a .cpp file I would define it and that would be its existence. This is how I would implement it:

```
typedef accel_namespace::Accelerometer&lt;Timer2&gt; accel;
DoSomething&lt;accel&gt; example_instance;
```

I wouldn't be able to see the variable I mentioned earlier and I would be able watch this variable either:

`TAccel accel`

Does that make sense?

haneefdm, 2020-01-27:
Oh, never mind. I did not notice the static part...

You can do an `arm-none-eabi-objdump --syms <elf-file-name>` to see what objects are actually available. This is what Cortex-Debug sees as valid symbols. Could you post the output, please. I am working in this area of having to do with static variables. Some variables don't even appear in the Variables window. See Issue #179. This example you have could be relevant. objdump also had (de)mangling options is the output was unreadable.

It is strange to have a static variable inside a templated class :-). Of course, it is not illegal, but I doubt you will get that static variable anywhere in the object file(s) unless you instantiate that class at least once or it would be like you never even had it. I doubt any code will be generated unless the class is instantiated at least once to create a concrete type.

Here is a topic that may be of interest.
https://www.fayewilliams.com/2014/12/23/static-variables-in-template-classes-and-methods/

I have a feeling that Cortex-Debug does not fully handle all the variants of C++ generated symbol types. Simple full examples would be great to see what is going on. I can volunteer to fix issues as I see them reported. I don't have any C++ programs myself that run on any embedded stuff.

senorclean, 2020-01-28:
Alright so this was the output from that command:

`200057cc  w    O .bss	000000b6 _ZN8lis2dw128Lis2dw12I6Timer4E13rx_ppi_bufferE`

lis2dw12 <- namespace
Lis2dw12 <- template class for accelerometer
Timer4 <- Timer class that's used as a template parameter for accelerometer
rx_ppi_buffer <- static variable I was referencing

Using this command gave me human readable output though:

`arm-none-eabi-objdump -C --syms <elf-file-name>`

`200057cc  w    O .bss	000000b6 lis2dw12::Lis2dw12<Timer4>::rx_ppi_buffer
`

haneefdm, 2020-03-07:
There is a beta release available for download now. This issue should be fixed. Please see the changelog. If you have additional issues  and I am sure will find new ones with C++, we should file separate issues

https://github.com/Marus/cortex-debug/releases/tag/v0.3.5-beta1

closed 2020-03-07 by haneefdm