Marus/cortex-debug #238: Github releases stopped at v0.2.7

Beetix, 2020-01-29:
Hi,

Is there a reason why releases on Github stop at v0.2.7 (vs v0.3.4 on VS Markplace)?

I am asking this because I am trying to download the latest release automatically through a script. I also have a bot setup to alert me on new releases on Github repos.

Thanks

haneefdm, 2020-01-29:
It was a mistake/oversight. We discussed this and agreed that future releases will be there.

Beetix, 2020-01-29:
@haneefdm Great! Thanks for the super fast reply.

haneefdm, 2020-01-29:
I will close this issue when our next release appears properly (hopefully).

haneefdm, 2020-03-07:
I just released a beta release of v0.3.5. I will follow that process and I am sure Marus will do as well and we should be in sync. My flow is,

* Make a github release first (draft form until satisfied)
* Publish github release & keep doing above for pre/alpha/beta releases. Only users watching the repo will be notified
* Finally: Publish the extension to VSCode market place & publish the github release. I think `vsce` has a way to do this in one step but I have not tried it. All VSCode users will be notified.

Any refinements, more automation suggestions are super welcome. Closing this for now.

closed 2020-03-07 by haneefdm