Marus/cortex-debug #239: Ability to specify RTT telnet port or arbitrary server args

sebastian-bartkowiak, 2020-01-31:
Hi.

I really like your extension. I think it lacks one major function. I am developing a system where two devices talk to each other, so my setup consists of two simultaneously debugged devices. The thing is - `JLinkGDBServerCL` which is my debug server by default uses fixed port for RTT communication with debugged devices, so with current setup I am able to use RTT only on one of the debuggers (the one that was started first). There is, however `-RTTTelnetPort` switch for the server CLI that forces it to use specified port. Unfortunately I am unable to force my `launch.json` to properly pass that arg when starting server. It would be great to give an option to either:
- specify `RTTTelnetPort` directly with dedicated option in launch.json, or
- give an ability to pass arbitrary args to server (with an option like `serverArgs`, exactly like `debuggerArgs`)

Thanks!

haneefdm, 2020-01-31:
There is an option in `launch.json` where you can specify an arbitrary gdb-server. Now you can do whatever you want, but starting the server (locally or remotely) is now your responsibility. I used it a little and it works but I have not tried with J-Link (don't see why not)
```JSON
      "servertype": "external",
      "gdbTarget": "localhost:2222",
```
You can also control for the most part what commands are sent to gdb/gdb-server at startup. See ChangeLog for some details

sebastian-bartkowiak, 2020-01-31:
I believe that my solution is simpler - allowing user to give arbitrary arguments while starting server. I threw together a PR for that, that solves this issue and seems to work (at least in my configuration).

PR #240 (2020-02-02): Fix issue #239
Fixes #239
closed 2020-02-02 by haneefdm