Marus/cortex-debug #241: gdb issue on Ubuntu 18.04

FloVanGH, 2020-02-03:
I can't use the extension with gdb on Ubuntu 18.04. On Ubuntu 18.04 the gdb executable is named `gdb-multiarch` and not `...-gdb`. I can only change the prefix to `gdb-mutliarch` but then the whole call is changed to `gdb-mutlicarch-gdb`. It would help if you could set the whole executable name instead of the prefix https://github.com/Marus/cortex-debug/blob/c93336bd2977b20e603af2cabb7dddd5043c8a0f/src/gdb.ts#L244.

haneefdm, 2020-02-03:
It is not just gdb, we also need objdump.

Could I ask where you are downloading the Ubuntu gdb releases from? We normally get them either from the chip vendor tools or https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Are you not developing for ARM Cortex devices? If not, I am not sure what else could break.

You could use symlinks I suppose.

FloVanGH, 2020-02-03:
With `apt get install gdb-multiarch`. Thank you for the hint with the symlink.

closed 2020-02-03 by haneefdm