Marus/cortex-debug #244: Remote debugging

mdx88, 2020-02-05:
I'm looking for a solution for remote debugging. Have founded option "gdbTarget" which could imply that this extension supports GDB servers running on machine other than localhost, however it still requires JLinkGBDServer to be installed on local machine. Even after installing JLink tools the extension is failing on local JlinkGDBServer initialisation phase.

Is there any option to make this extension to work for remote debugging? 

Here is config I use:
``` javascript
"name": "Cortex Debug",
"cwd": "${workspaceRoot}",
"executable": "${workspaceRoot}\\Temp\\test.elf",
"request": "launch",
"type": "cortex-debug",
"device": "EFM32GG11B820F2048",
"interface": "swd",
"servertype": "jlink",
"serverpath": "C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServer.exe",
"armToolchainPath": "${workspaceFolder}\\Tools\\gcc-arm-none-eabi-win32\\bin",
"gdbTarget": "192.168.1.46:2331",
```

haneefdm, 2020-02-05:
For using an arbitrary gdb-server you use the `"servertype": "external"` and then specify `"gdbTarget"` (which is only used for an external-server). The gdb server can be anything, running on any machine on your network. You first start the server and then launch/attach and Cortex-Debug will connect to it via the specified port. If it is working, please let us know and close this issue.

mdx88, 2020-02-05:
It works!
Thank you.

closed 2020-02-05 by mdx88

GitMensch, 2021-09-17:
Does this mean one could use external + gdbTarget to debug _any_ environment where a gdbserver runs (including C running on WSL2, remote Linux Server, ...) as long as the configured GDB supports the target (powerpc, raspi, x86, ...)?
If not: what would be missing to work without jlink/openocd?

haneefdm, 2021-09-17:
@GitMensch I would not say for sure without testing but generally yes. There are more dependencies on 32-but-ness (register values) and endian-ness (memory viewer) than the processor type. So long as gdb and the gdb-server behave, you are good to go.

I could not say for sure what would not work. The only way to find out is .... try.

GitMensch, 2021-09-17:
Why has the memory viewer an endian-ness relation? If you use memory objects to access the memory on the GDB side you just get it as-is, and if you want to you could convert that to a byte array which can then be accessed via GDB mi. But I don't know the parts where the extension currently does this. Maybe it is reasonable to create a new issue for that part?

I'm not sure: Do you use the hexedit extension with the logical "read only file" that you've implemented there for the UI part (that one as endianness-options on its own)?

haneefdm, 2021-09-17:
@GitMensch No, the memory viewer doesn't care. I misspoke. I know there is some dependency somewhere. SWO/RTT care.

I have a prototype of the integrated hexedit version -- I fixed/enhanced it to make it work. I was all ready to deploy and all of a sudden their UI changed drastically and I am not sure it is for the better. There is a lot of flicker and it interferes with normal workflow (editing, debugging, source control, etc.). I don't want the memory viewer to be the center of the universe. I  periodically check to see if it is ready.

VSCode proper is also making changes to the read-only mode and is not yet stabilized. I think it is usable now.