Marus/cortex-debug #249: marus25.cortex-debug-0.3.4/support/gdbsupport.init:120: Error in sourced command file:

perlindgren, 2020-02-14:
Hi

I ran into this error. Not sure if its self inflected, I've been playing around with compiling vscode extensions locally, so I might have messed something up. 

I'm on latest vscode (stable), and on latest cortex-m-debug (stable/release/installed through the marketplace).
``` shell
I get this error (in the DEBUG CONSOLE):
Launching server: "openocd" "-c" "gdb_port 50000" "-s" "/home/pln/rust/e7020e_2020/e7020e_2020" "-f" "interface/stlink-v2-1.cfg" "-f" "target/stm32f4x.cfg"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /home/pln/rust/e7020e_2020/e7020e_2020/target/thumbv7em-none-eabihf/debug/app...
/home/pln/.vscode/extensions/marus25.cortex-debug-0.3.4/support/gdbsupport.init:120: Error in sourced command file:
Ambiguous command "def EnableDWTSync": define, define-prefix.
```
My setup:

``` shell
openocd --version
Open On-Chip Debugger 0.10.0+dev-01060-g0a11537b3 (2020-02-14-21:22)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```
``` shell
 arm-none-eabi-gdb --version
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

My launch:
``` json
        // Launch configuration for `app`
        // - debug
        // - semihosting
        // - run to main
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "name": "app (debug)",
            "preLaunchTask": "cargo build",
            "executable": "./target/thumbv7em-none-eabihf/debug/app",
            "configFiles": [
                //"interface/stlink.cfg",  // deprecated setup script
                "interface/stlink-v2-1.cfg",
                "target/stm32f4x.cfg"
            ],
            "postLaunchCommands": [
                "monitor arm semihosting enable"
            ],
            "runToMain": true,
            "cwd": "${workspaceRoot}"
        },
```

My task
``` json
        {
            "type": "shell",
            "label": "cargo build",
            "command": "cargo build",
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "problemMatcher": [
                "$rustc"
            ]
        },
```

mozram, 2020-02-16:
Exactly issue I facing right now with 
`Ambiguous command "def EnableDWTSync": define, define-prefix.`

haneefdm, 2020-02-16:
You are using `gdb 9.1`? This has not been validated yet and not even supported by ARM AFAIK. Also, that file `~/.vscode/extensions/marus25.cortex-debug-0.3.4/support/gdbsupport.init' has not changed in two years and not sure why gdb 9.1 is unable to parse it. Sounds like a bug in gdb.

The ARM supported ones only go to 8.3.x (see release notes in page below)

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

You can use command-line `arm-none-eabi-gdb` and simply do a 'source gdbsupport.init' and it should work. If not, it is definitely a bug/incompatibility with 9.1 and should be reported to the gdb team ASAP. All that file does is define a bunch of gdb functions/macros

haneefdm, 2020-02-16:
Ah, I see, that is the first line that uses `def` shorthand/alias vs. the rest of the file above uses `define` to define a macro.

Could one of you guys search-replace `'def '` with `'define '` and see if the error goes away? If that is indeed the problem, I can fix it for the next release coming up very soon. I am on vacation for the next three days but if you can get back quickly, then I can do something about it. I don't want to touch a file that hasn't been touched in ages.

It should still be reported to the gdb team as this can cause many scripts to fail. We should not be using shorthands/aliases either but... (like `bt` for `backtrace` or `b` for `break`). Looks like they removed `def` as an alias for `define`. Wonder what else they did! `gdb 9` is a major rev change so...

About using `gdb 9.1`, you are really on your own :-) Expect more problems.

haneefdm, 2020-02-16:
Hmm, I can't tell what is a pre-defined alias and what is an actual command in gdb.

`help aliases` is no use either because it only lists a very small subset. According to the following, `ni` is an alias but `def` appears to be a real command that they removed.

```text
$ arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

$ arm-none-eabi-gdb
...
(gdb) help ni
Step one instruction, but proceed through subroutine calls.
Usage: nexti [N]
Argument N means step N times (or till program stops for another reason).

(gdb) help def
Define a new command name.  Command name is argument.
Definition appears on following lines, one command per line.
End with a line of just "end".
Use the "document" command to give documentation for the new command.
Commands defined in this way may accept an unlimited number of arguments
accessed via $arg0 .. $argN.  $argc tells how many arguments have
been passed.

(gdb) help define
Define a new command name.  Command name is argument.
Definition appears on following lines, one command per line.
End with a line of just "end".
Use the "document" command to give documentation for the new command.
Commands defined in this way may accept an unlimited number of arguments
accessed via $arg0 .. $argN.  $argc tells how many arguments have
been passed.
(gdb) 
```

haneefdm, 2020-02-16:
I updated the `gdbsupport.init` file and it works as before with gdb 8.3.x I don't want to play with gdb 9.1 yet.

You can see my commit here https://github.com/Marus/cortex-debug/pull/248/commits/0fa74869d3977a5d0bb5b75fbb921c75389d1544

My PR is still open but... I want to just close this issue. You have a workaround and the fix will be in the next release.

closed 2020-02-16 by haneefdm

perlindgren, 2020-02-18:
Can confirm the patch works for 9.1. Thanks for the quick fix.

Luxni, 2020-03-06:
I get the bug in arm-none-eabi-gdb 9.1.
 and just replace all the "def " with "define " fix it.

haneefdm, 2020-03-06:
@Luxni Not sure if it is a question or a statement. Yes, replacing all the `'def '` to `'define '` will get you moving until the next release which is due very soon