Marus/cortex-debug #253: F5 can enter debug, click single step to run and cannot jump

xmlwancl, 2020-02-23:
Cortex-Debug 0.3.4    openocd  0.10.0
Debug into assembly：
`Reset_Handler:  
  ldr   sp, =_estack     /* set stack pointer */   Pause here

    `
Click continue  Can't continue run 
MCU Can't run
launch.json
"configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "build/cubetest.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "svdFile": "tools/STM32F401X.svd",
            "configFiles": [
                "tools/stlink-v2.cfg",
                "tools/stm32f4x.cfg",
            ]
        }
    ]

haneefdm, 2020-02-23:
Not the right answer for you but, could you try the following.

In your `launch.json`, enable `"runToMain"`. See if that works. I noticed that OpenOCD has issues at startup, just after reset with some devices. Since continue did not work for you, I doubt this will work either, but worth a try. Also, enabled `showDevDebugOutput` in `launch.json` to see more verbose output

Check the Output tab (Debug Adapter section) to see messages from OpenOCD. Please provide messages from both the `Debug Console` and the `Output Tab`

More advanced troubleshooting: I don't recommend this, but it is the final resort. Use command-line gdb and openocd to see what is going on. If that does not work, neither will Cortex-Debug

xmlwancl, 2020-02-23:
output：
pen On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v31 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.252337
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10076413
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0xa029aca0 msp: 0x11748d2c
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0xa029aca0 msp: 0x11748d2c
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xd0f74546 msp: 0x11748d08
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0x0f00f1b8 msp: 0x8000f8d0
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0x0f00f1b8 msp: 0x8000f8d0

Debug Console:
pen On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v31 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.252337
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10076413
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0xa029aca0 msp: 0x11748d2c
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0xa029aca0 msp: 0x11748d2c
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xd0f74546 msp: 0x11748d08
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0x0f00f1b8 msp: 0x8000f8d0
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 0x0f00f1b8 msp: 0x8000f8d0

haneefdm, 2020-02-23:
I have a feeling that programming/reset is failing. Also, a hard fault which is usually an infinite loop. A few hints from output below
```
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xd0f74546 msp: 0x11748d08
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
```

closed 2020-02-23 by xmlwancl