Marus/cortex-debug #255: configFiles with environment variable not supported

ziyouchutuwenwu, 2020-03-07:
Hi, my launch.json is like this
```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "openocd_debug",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/rt-thread.elf",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "device": "stm32f411ce",
      "gdbpath": "/usr/bin/gdb-multiarch",
      "configFiles": [
        "/home/`whoami`/dev/embedded/stm32/openocd/stm32f4x/jlink_board.cfg"
      ],
      // "preLaunchTask": "build",
      "runToMain": true
    }
  ]
}
```

for the configFiles section, i need to input the absolute path, can i use ~ or $HOME or other variable instead of absolute path?
great thanks for this amazing plugin, i can debug happiliy under linux.

thank you very much!!!!!!!!!!

haneefdm, 2020-03-07:
I don't believe those strings are evaluated like on a shell command line. `~` may not work either as that is shell specific.

But, VSCode, lets you use things like `${workspaceRoot}` for predefined VSCode variables or `${env:HOME}` for any environment variables. The following link may be helpful

https://code.visualstudio.com/docs/editor/debugging

closed 2020-03-07 by haneefdm

ziyouchutuwenwu, 2020-03-07:
> I don't believe those strings are evaluated like on a shell command line. `~` may not work either as that is shell specific.
>
> But, VSCode, lets you use things like `${workspaceRoot}` for predefined [...]

thank you very much! ${env:HOME} works for me~