Marus/cortex-debug #257: question: is there a way to log the raw commands being sent to start gdb, etc?

busticated, 2020-03-13:
hey again :wave:

congrats on the `v0.3.5` release - looks like a bunch of nice additions and fixes :clap::pray:

from time to time i find myself wanting to see the underlying commands that are being sent to start `openocd` and `gdb` but to date i've been unsuccessful in doing so. i'm wondering if there's a debugging option already in place to log those out to VSCode's built-in dev-tools console or elsewhere?

thanks 👍

busticated, 2020-03-13:
wait...  🤦‍♂ lol... looks like they are printed out at the top of the debug console upon launching - yay 🎉 :+1:

closed 2020-03-13 by busticated

haneefdm, 2020-03-13:
Thanks, I added that several months ago :-). There is also `showDevDebugOutput`, when enabled you will see all the chatter between Cortex-Debug and gdb. That was always there. What is new with `v0.3.5` is that you will now see relative timestamps in case something is taking too long.