Marus/cortex-debug #258: 'Unable to get thread information' when setting breakpoint

JojoS62, 2020-03-14:
I'm getting the following message:
```
Unable to get thread information: Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again. (from thread-info 1)
```
This happens when I set a breakpoint in an ISR, but the execution is stopped at another ISR. Debugging on a STM32F4, running some code with mbed-os. Debugprobe is a BMP
cortex-debug is 0.3.5.

haneefdm, 2020-03-14:
In your launch.json can enable debugging output and see what it prints in the Debug Console `"showDevDebugOutput": true` 

One thing I remember is that in a Handler, the stack pointer is different and you cannot see a true stack unless you fix it. Some gdb servers handle this better than others. I don't have any experience with `BMP` and how it behaves in an ISR/FaultHandler. Basically, if gdb can't figure it out, neither can Cortex-Debug.

@Marus is better at this than I am.

michaelkollmann, 2020-03-29:
I am having the same issue so here is some more information on the problem.

After activating the "showDevDebugOutput" option, starting the debug process, clearing the output window and setting a break point the following is being printed to the output window

```
0001018144: 49-exec-interrupt
0001018150: GDB -> App: {"token":49,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0001018168: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"\nProgram"}]}
0001018168: 
Program
0001018168: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":" received signal SIGTRAP, Trace/breakpoint trap.\n"}]}
0001018169:  received signal SIGTRAP, Trace/breakpoint trap.
0001018170: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"nrf_pwr_mgmt_run () at C:/Repos/PROJECTFOLDER/sdk/components/libraries/pwr_mgmt/nrf_pwr_mgmt.c:361\n"}]}
0001018170: nrf_pwr_mgmt_run () at C:/Repos/PROJECTFOLDER/sdk/components/libraries/pwr_mgmt/nrf_pwr_mgmt.c:361
0001018170: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"361\t        __SEV();\n"}]}
0001018170: 361	        __SEV();
0001018171: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["reason","signal-received"],["signal-name","SIGTRAP"],["signal-meaning","Trace/breakpoint trap"],["frame",[["addr","0x0002b106"],["func","nrf_pwr_mgmt_run"],["args",[]],["file","C:/Repos/PROJECTFOLDER/sdk/components/libraries/pwr_mgmt/nrf_pwr_mgmt.c"],["fullname","C:\\Repos\\PROJECTFOLDER\\sdk\\components\\libraries\\pwr_mgmt\\nrf_pwr_mgmt.c"],["line","361"]]],["thread-id","1"],["stopped-threads","all"]]}]}
0001018171: 50-break-delete 4
0001018175: 51-thread-list-ids
0001018176: 52-data-list-register-values N
0001018181: GDB -> App: {"token":50,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0001018181: 53-break-insert "c:\\Repos\\PROJECTFOLDER\\src\\app\\main.c:93"
0001018181: 54-break-insert "c:\\Repos\\PROJECTFOLDER\\src\\app\\main.c:96"
0001018182: GDB -> App: {"token":51,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
0001018182: 55-thread-info 1
0001018196: GDB -> App: {"token":52,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","0"],["value","0"]],[["number","1"],["value","3758153984"]],[["number","2"],["value","0"]],[["number","3"],["value","0"]],[["number","4"],["value","0"]],[["number","5"],["value","0"]],[["number","6"],["value","268439552"]],[["number","7"],["value","536936408"]],[["number","8"],["value","0"]],[["number","9"],["value","0"]],[["number","10"],["value","536870912"]],[["number","11"],["value","0"]],[["number","12"],["value","765"]],[["number","13"],["value","0x2000ffd8"]],[["number","14"],["value","176375"]],[["number","15"],["value","0x2b106 <nrf_pwr_mgmt_run+30>"]],[["number","25"],["value","1627389952"]],[["number","91"],["value","536936408"]],[["number","92"],["value","0"]],[["number","93"],["value","0"]],[["number","94"],["value","0"]],[["number","95"],["value","0"]],[["number","96"],["value","4"]],[["number","97"],["value","0"]],[["number","98"],["value","0"]],[["number","99"],["value","0"]],[["number","100"],["value","0"]],[["number","101"],["value","0"]],[["number","102"],["value","0"]],[["number","103"],["value","0"]],[["number","104"],["value","0"]],[["number","105"],["value","0"]],[["number","106"],["value","0"]],[["number","107"],["value","0"]],[["number","108"],["value","0"]],[["number","109"],["value","0"]],[["number","110"],["value","0"]],[["number","111"],["value","0"]],[["number","112"],["value","0"]],[["number","113"],["value","0"]],[["number","114"],["value","0"]],[["number","115"],["value","0"]],[["number","116"],["value","0"]],[["number","117"],["value","0"]],[["number","118"],["value","0"]],[["number","119"],["value","0"]],[["number","120"],["value","0"]],[["number","121"],["value","0"]],[["number","122"],["value","0"]],[["number","123"],["value","0"]],[["number","124"],["value","0"]],[["number","125"],["value","0"]],[["number","126"],["value","0"]],[["number","127"],["value","0"]],[["number","128"],["value","0"]],[["number","129"],["value","0"]],[["number","130"],["value","0"]],[["number","131"],["value","0"]],[["number","132"],["value","0"]],[["number","133"],["value","0"]],[["number","134"],["value","0"]],[["number","135"],["value","0"]],[["number","136"],["value","0"]],[["number","137"],["value","0"]],[["number","138"],["value","0"]],[["number","139"],["value","0"]],[["number","140"],["value","0"]],[["number","141"],["value","0"]],[["number","142"],["value","0"]],[["number","143"],["value","0"]],[["number","144"],["value","0"]],[["number","145"],["value","0"]]]]]}}
0001018197: GDB -> App: {"token":53,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","5"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x0003054a"],["func","main"],["file","C:/Repos/PROJECTFOLDER/src/app/main.c"],["fullname","C:\\Repos\\PROJECTFOLDER\\src\\app\\main.c"],["line","93"],["thread-groups",["i1"]],["times","0"],["original-location","c:\\Repos\\PROJECTFOLDER\\src\\app\\main.c:93"]]]]}}
[... cut ...]
            "interface": "swd",
            "armToolchainPath": "C:/gcc-arm-none-eabi/7-2018-q2-update/bin/",
            "runToMain": true,
            "svdFile": "./sdk/modules/nrfx/mdk/nrf52.svd",
            "showDevDebugOutput": true,
            "preLaunchTask": "Launch RTT Viewer"
        }
    ]
}
```

For now the error is just annoying because everything works as it should but I'd greatly appreciate if this could be fixed.
Except for that Cortex-Debug is great and does exactly what it should! 👍

haneefdm, 2020-03-30:
The issue should be addressed by V0.3.6.

closed 2020-03-30 by haneefdm

JojoS62, 2020-04-08:
yes, V0.3.6 fixed this issue. Thanks!