Marus/cortex-debug #260: Unable to load Serial Port Module after VScode update

diegoherranz, 2020-03-18:
VSCode has updated Node.js so some work required on the modules again, similar to https://github.com/Marus/cortex-debug/issues/243

Thanks!

```
Version: 1.43.0
Commit: 78a4c91400152c0f27ba4d363eb56d2835f9903a
Date: 2020-03-09T19:44:52.965Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 4.4.0-174-generic
```

haneefdm, 2020-03-18:
Darn it, we just made a release and the Node.js version changed again. Could you try the following in a shell for me while I prepare another release?

```bash
cd ~/.vscode/extensions/marus25.cortex-debug-0.3.5/
./serial-port-build.sh 7.1.11 12.8.1
rm -fr ./tmp
```

The serial port module should work again. Please report back.

diegoherranz, 2020-03-18:
I've tested it and it worked perfectly. Nice thing that that script was added!

Is it just a matter of committing the 12.8.1 folder created after running this script?
If so I can help submitting a PR for the Linux case at least.

Thanks!

haneefdm, 2020-03-18:
Sure, submit a PR. I am glad it worked!!! I will do it for the other platforms.

diegoherranz, 2020-03-19:
Done. I haven't removed the v12.4.0 folder for linux x64 since I guess some users may still be using a previous version. What is the policy here? Leave the last 2 Node.js versions?
Thanks!

Marus, 2020-03-20:
Typically try to keep the current version and previous version around -
don't keep any more than that to avoid bloating too much. And thanks for
doing this - always great to get active help from the community.

diegoherranz, 2020-03-20:
No problem

haneefdm, 2020-03-30:
Should work with V 0.3.6

closed 2020-03-30 by haneefdm