Marus/cortex-debug #261: Breakpoint not triggering

leocb, 2020-03-18:
When the debug starts, i get the following errors and breakpoints are not triggered
If I hit pause I can see the callstack, so I assume the debugger is working.

This is a STM32F103 project

[image]

haneefdm, 2020-03-18:
Bit more info please? I am not even sure you are using the Cortex-Debug debugger. MS has their own generic one.

Please show us your `launch.json` and which config you were using.

leocb, 2020-03-18:
Sure!
[image]

haneefdm, 2020-03-18:
Hmmm. The class `RawDebugSession` doesn't even exist in our code. I believe that is a generic MS cpptools class name.

VSCode has not even used this extension, it appears. But, you are using all the latest stuff. This may be a super-recent bug in VSCode itself where it is not activating our extension.

Could you totally quit all instances of VSCode and start over. Something in extension loading is missing.

leocb, 2020-03-18:
Completely restarted VSCode and windows, same problems

haneefdm, 2020-03-18:
Confused as hell. Could you post what you are seeing in the Debug Console?

leocb, 2020-03-18:
here ya go
https://pastebin.com/b5VDRjX7

leocb, 2020-03-18:
VSCode just released a update and I also updated my GDB to 9 2019-4-major

still broken

haneefdm, 2020-03-18:
At this point, I think it is COVID-19.  I have been awake for a super long time. I will investigate a bit later. Also, let the new release of VSCode marinate a bit.

leocb, 2020-03-18:
hahahaha figures that would be the cause!

OK, no problem! thanks for the attention :)

haneefdm, 2020-03-18:
Okay, maybe some hints....

```text
0000001206: undefinedC:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
```
See #199

```text
0000001219: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from C:\\Users\\HWSolinftec\\Documents\\GitLab\\Firmware\\0000_TagCarreta2\\Tests/build/TagCarreta_TT_Test.elf...\n"}]}
0000001219: Reading symbols from C:\Users\HWSolinftec\Documents\GitLab\Firmware\0000_TagCarreta2\Tests/build/TagCarreta_TT_Test.elf...
0000001221: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"(No debugging symbols found in C:\\Users\\HWSolinftec\\Documents\\GitLab\\Firmware\\0000_TagCarreta2\\Tests/build/TagCarreta_TT_Test.elf)\n"}]}
0000001221: (No debugging symbols found in C:\Users\HWSolinftec\Documents\GitLab\Firmware\0000_TagCarreta2\Tests/build/TagCarreta_TT_Test.elf)
```
This may be a bigger problem. GDB has issues with backslashes and I remember us passing gdb stuf with forward slashes. It says it cannot find any debug symbols. Makes no sense because later on, I see it does know what symbols exist.

leocb, 2020-03-20:
Hi, sorry for the delay
thanks for the tip, I created the HOME and USERPROFILE env vars and the warning went away
I saw those inconsistent back/forward slashes too. I change the paths on the launch.json to use relative paths, still broken.

here's my current launch.json:

```JSONC
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [{
    "type": "cortex-debug",
    "request": "launch",
    "servertype": "openocd",
    "cwd": "${workspaceFolder}",
    "executable": "./build/TagCarreta_TT_Test.elf",
    "name": "Debug (OpenOCD)",
    "device": "STM32F103xE",
    "showDevDebugOutput": true,
    "configFiles": [
      "./.vscode/stlink.cfg",
      "./.vscode/stm32f1x.cfg"
    ]
  }]
}
```

haneefdm, 2020-03-20:
I am glad it worked out for you. I still don't understand why it was failing

leocb, 2020-03-20:
I might have expressed myself badly in the previous comment, **it's still not working**, it's just the `warning: Couldn't determine a path for the index cache directory.` that went away 😶

leocb, 2020-03-23:
Related to #263 ?

haneefdm, 2020-03-25:
I don't think so. If breakpoints are not being hit

- PC never got there, code got optimized out
- gdb could not set the requested breakpoints. To test this, try doing `info breakpoints` in the Debug Console

haneefdm, 2020-03-30:
@leocb  Could you please try the newest release V0.3.6 and report back.

leocb, 2020-03-30:
Hi, just tested it, same problems. Same settings, Cortex debug @ 0.3.6 - VSCode @ 1.43.2

Here's the notification: 
[image]

And the debug console dump:
https://pastebin.com/Nm7UTrQJ

haneefdm, 2020-03-30:
I think you still have the same problem. It is there at the beginning of the debug console dump... Your executable seems to be an invalid elf file or not an elf file at all or all the symbols have been stripped. Are you sure you are compiling/linking the executable for debug? (gcc -g)

look at the last line below

```text
0000003395: 7-enable-pretty-printing
0000003734: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
0000003735: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from C:\\Users\\HWSolinftec\\Documents\\GitLab\\Firmware\\0000_TagCarreta2\\Tests\\build\\TagCarreta_TT_Test.elf...\n"}]}
0000003735: Reading symbols from C:\Users\HWSolinftec\Documents\GitLab\Firmware\0000_TagCarreta2\Tests\build\TagCarreta_TT_Test.elf...
0000003737: GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"(No debugging symbols found in C:\\Users\\HWSolinftec\\Documents\\GitLab\\Firmware\\0000_TagCarreta2\\Tests\\build\\TagCarreta_TT_Test.elf)\n"}]}
0000003737: (No debugging symbols found in C:\Users\HWSolinftec\Documents\GitLab\Firmware\0000_TagCarreta2\Tests\build\TagCarreta_TT_Test.elf)
```
No symbols mean no breakpoints . launch gdb from command-line. If gdb is not working, neither can this extension.

you can also run `objdump --syms` on the elf file to see what symbols you get, if any.

leocb, 2020-03-30:
Thanks for your help!

I checked my makefile, the debug version of the binary was not using `-g`, just the `-Og` optimization level flag. I guess I just inserted the `O` in there instead of adding another whole flag.

odd thing is, I've been using this same "base" makefile on other projects, this was the first one that didn't link the debug symbols. weird.

Thank you so so much for your time! this is solved now 😄

closed 2020-03-30 by leocb

leocb, 2020-03-30:
OK, this is solved, but I'm very confused now, in my modified STM makefile I have this line:

```makefile
debug: DEBUG = 1
debug: OPT = -Og
```

this is so I can call make debug or make final
Then a little further down the file there're these:

```makefile
CFLAGS = $(MCU) $(C_DEFS) $(C_INCLUDES) $(OPT) -Wall -fdata-sections -ffunction-sections -fno-exceptions

ifeq ($(DEBUG), 1)
CFLAGS += -g -gdwarf-2
endif
```

The lines above checks if the `DEBUG` var is set to 1 and then add the `-g` flag. I did set the var when I'm compiling the debug target, so why won't it work?

I know this is not "on topic" anymore, but if you could explain this behavior to me I'd appreciate it very much.

Thanks again 😄

haneefdm, 2020-03-30:
Makefiles are more an art than science :-) They are magical/wonderful as well as frustrating.

tdjastrzebski, 2020-05-01:
@haneefdm I would say more frustrating. To the point I built my own build runner [cppbuild](https://github.com/tdjastrzebski/cppbuild)