Marus/cortex-debug #265: Target not initialized properly when servertype is external

bscramps, 2020-03-25:
I am working with an STM32F756, debugging with a J-Link. My dev tools live in WSL, so I need to connect to the J-Link via Windows, since WSL software cannot directly talk to the J-Link.

I have accomplished this two ways:
Method 1:
- Windows runs J-Link Remote Server
- Cortex-debug configured to launch a JLinkGDBServer in WSL, passing appropriate options to connect to the J-Link via localhost.

Method 2 - The subject of this issue:
- Windows runs JLinkGDBServerCL
- Cortex-debug configured with servertype=external and ipAddress=localhost 

Method 1 works, but sometimes something gets "stuck" and I am unable to connect to the J-Link for some time. I suspect that this is a connection left open when the debug launch fails. It seems that the Segger software gets confused. 

Method 2 is what I have been testing with recently. This method will consistently connect to the J-Link and the target, but it seems that not all of the correct commands are sent to the JLinkGDBServer to appropriately reset the target. This is a suspicion at this point, and I'm unsure what to test next. The evidence that leads me to this hunch are seemingly random core register values at startup; sometimes PRIMASK=1, someitmes not. Sometimes XPSR->ISR_NUMBER=3 (indicating Hard Fault). 

To my knowledge there is no practical reason that an ARM core, properly reset, would startup in an ISR context other than Reset, nor have PRIMASK set to 1.

haneefdm, 2020-03-25:
You can enable debug output in your `launch.json` and see what exact commands are being sent in the Debug Console. Bit tedious but only way to debug and a bit of a learning curve on how GDB works.
```JSON
"showDevDebugOutput": true
```
Btw, you can override almost every command sent to the server via gdb int `launch.json`

closed 2020-03-31 by haneefdm

bscramps, 2020-04-06:
Needed to add to launch.json: "postLaunchCommands": ["monitor halt", "monitor reset"],