Marus/cortex-debug #266: Another "JLink debugger not starting" issue

aakre, 2020-03-25:
Hi all!

Long time user of the extension (love it!). Now I've just switched laptop and tried to make my project compile and run on target STM32L462VE, but the debugger silently exits with no trace or error.
if I start debugging with my JLink unplugged from the computer, I get a pop-up in vscode with "Failed to launch undefined GDB server:Timeout". Hmmm... mysterious "undefined"? It seems like vscode can start the GDB server though (see adapter output from when it is plugged in). 
I have exact same version of JLink and vscode on two systems, one is working, one is not, so I hope we can figure this out :) 

vscode dump new system:
Version: 1.43.2
Commit: 0ba0ca52957102ca3527cf479571617f0de6ed50
Date: 2020-03-24T07:52:11.516Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 5.3.0-7642-generic snap

cortex-debug: 0.3.5
J-Link: 6.42e / 6.64b
host: Pop!_OS 19.10

**Adapter output when pluged in**
SEGGER J-Link GDB Server V6.64b Command Line Version
JLinkARM.dll V6.64b (DLL compiled Mar 20 2020 10:08:28)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32L462VE
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32L462VE
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Mar 18 2020 17:38:51
Hardware: V10.10
S/N: 50123580
Feature(s): GDB
Checking target voltage...
Target voltage: 3.34 V
Listening on TCP/IP port 50000
Connecting to target...

haneefdm, 2020-03-25:
Not really sure what your question is, but...

There is a timeout of 20 seconds in Cortex-Debug. We are waiting for a string `Waiting for GDB connection` from JLink server, I don't see that. Cortex-Debug will not start until that actually happens. What changed? Not sure.

You can override that regex in `launch.json` with the `overrideGDBServerStartedRegex` property.

But what you attached does not look like actual JLink output. I haven't used it in a while. Can I see the actual JLink server output?

aakre, 2020-03-25:
thanks for answering :) 
Sorry for being vague. The problem is that I'm unable to start a debug session and I don't understand why. I've been working on a project for the past 2 years using vscode and cortex-debugger. Now I have a new machine and has set it up as before. When I start debug from vs code it automatically stops after 1-2 seconds. No error message. I need help to figure out what goes wrong :)

1. What should I set overrideGDBServerStartedRegex to to override the regex? 
2. How should I produce the actual JLink output? I copied the output from "Adapter output" in vscode

aakre, 2020-03-25:
Just an update. I tried to start JLinkGDBServer manually in a shell using the same command. Then I get to "Waiting for GDB connection..."

```bash
./JLinkGDBServer -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32L462VE
SEGGER J-Link GDB Server V6.42e Command Line Version

JLinkARM.dll V6.42e (DLL compiled Feb 21 2019 15:26:31)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32L462VE
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32L462VE
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Mar 18 2020 17:38:51
Hardware: V10.10
S/N: 50123580
Feature(s): GDB
Checking target voltage...
Target voltage: 3.32 V
Listening on TCP/IP port 50000
Connecting to target...Connected to target
Waiting for GDB connection...
```

haneefdm, 2020-03-25:
May I see the literal output JLink server is giving you in the 'Adapter Output' section of the Output tab?

It should be **identical** to what you are seeing when you are running it manually. In your previous message, it stopped at `Connecting to target...` and nothing more.

aakre, 2020-03-25:
My original post contain the actual output from "Adapter Output" in vscode. It just stops there at "Connecting to target..." :/

haneefdm, 2020-03-25:
Are you using 'JLinkGDBServer' or JLinkGDBServer.CL. You will need the CL version (the command-line version)

No sir, you did not seem to attach the actual output, near as I can tell. Why does it not look the same as what is in VSCode? You did not attach what the Debug Console showed either. You are giving me selective information.

I am trying to help you. and you are getting testy. Sorry, the first time I ever got this upset. Maybe I missed something. Sorry.

aakre, 2020-03-26:
I think we got off on the wrong foot here :( I do not mean to be testy. English is not my native language so I guess I'm not able to express myself properly, sorry. 

Can we please start fresh? :)
I really appreciate you helping me! I've read your posts a couple of times now and I've tried to collect the information you're asking for. Please instruct me if there is something else you need :) 

1. I use the command line version of JLink
```bash
lrwxrwxrwx  1 oyvind oyvind       19 Mar 24 17:28 JLinkGDBServer -> JLinkGDBServerCLExe
```
2. I'm mainly using JLink 6.64b but in my earlier post where I ran JLinkGDBServer manually in a shell I actually used 6.42e. Try to ignore this. I get the same console output regardless of JLink version when I run manually from a shell. That is, when running from a shell I get to the line 'Waiting for GDB connection...', but when running from vscode I do not. So yes, something is different in the output when I run JLinkGDBServer from vscode. Seems like JLink is working fine, but something breaks along the way when starting from vscode.
3. Best attempt at literal JLink output. Okay, so here is everything from 'Adapter Output' section in 'Output' tab. The final line in the output really is 'Connecting to target...'. It stops there. If this is not the information you need, then please tell me how I can get it for you :) 
```bash
SEGGER J-Link GDB Server V6.64b Command Line Version
JLinkARM.dll V6.64b (DLL compiled Mar 20 2020 10:08:28)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32L462VE
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32L462VE
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Mar 18 2020 17:38:51
Hardware: V10.10
S/N: 50123580
Feature(s): GDB
Checking target voltage...
Target voltage: 3.34 V
Listening on TCP/IP port 50000
Connecting to target...
```

4. Here is output from Debug Console:
```bash
0000000008: Reading symbols from '/home/oyvind/fenalaar/com-board/build/main.elf'
0000000044: Finished reading symbols
0000000049: Please check OUTPUT tab (Adapter Output) for output from /home/oyvind/fenalaar/com-board/../sw/tools/JLink_Linux_V664b_x86_64/JLinkGDBServer
0000000049: Launching server: "/home/oyvind/fenalaar/com-board/../sw/tools/JLink_Linux_V664b_x86_64/JLinkGDBServer" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "STM32L462VE"
0000000306: Launching GDB: "/home/oyvind/fenalaar/sw/tools/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
0000000309: 1-gdb-set target-async on
```

5. Is there any other info I can provide for you? Can I enable verbose output somehow?

6. Could it be some system settings that prevents vscode+extension to use JLink properly? I mean the project structure with settings, tasks.json, launch.json etc is the same as I've been using for 1-2 years now, but this week I switched to a new laptop :) 

cheers!

haneefdm, 2020-03-26:
Thanks. It appears that the output from JLink server is not fully flushed. It may have finished connecting but has not flushed its stdout. Not sure how to debug this or simulate the behavior. I have a feeling the version of **JLink matters a lot**. I would try the same version as your older laptop if possible. There is buffering going on in Unix pipes and the writing end has to flush. **I would report this to SEGGER**. What you see interactively (in a tty/terminal/console) may not be the same as what occurs on a pipe.

Until we see the target string I [mentioned above](https://github.com/Marus/cortex-debug/issues/266#issuecomment-604091675), we cannot launch gdb. But we can change the target string though. Could you try the following in your `launch.json`
```JSON
    "overrideGDBServerStartedRegex": "Listening on TCP/IP port"
```
I will try to see if I can reproduce the problem with your version of JLink server, but that will take some time. I don't have any HW with me right now to test. Unfortunately, you are on a Linux and me a Mac so the behavior may not be the same either.

aakre, 2020-03-26:
Okay, so I tried JLink version V6.42e (same as on old laptop) and I added 
```json
 "overrideGDBServerStartedRegex": "Listening on TCP/IP port"
```
unfortunately it did not help :( 

The only difference I can spot is that now 'Adapter Output' is showing even less information. The final line is 'Listening on TCP/IP port 50000' (regardless of 'overrideGDBServerStartedRegex' setting).

I will reach out to Segger as well for help. Do you have any other ideas or tricks I can try in the meantime? :) 

cheers!

haneefdm, 2020-03-26:
That should work as far as Cortex-Debug is concerned. I see that gdb has been launched. gdb is not launched until the server is ready to accept a tcp/connection from gdb. Looks like JLink is literally stuck acquiring the target (your board/chip). No idea why it works from command-line.

You can also start JLink from command-line and try the following in your launch.json
```JSON
"servertype": "external",
"gdbTarget": 50000, // This should be whatever JLink prints as tcp/ip port
```
Above is not a great solution, but try it.

To enable verbose output in debug console add `"showDevDebugOutput": true`. Looks like you already did this since I see timestamps.

aakre, 2020-03-26:
Thank you, you put me on the track to find the solution :D

Turns out my version of arm-none-eabi-gdb depends on libncurses5 whilst my system has version 6 installed. Can't believe I did not find this sooner, I thought I tried to run arm-none-eabi-gdb manually yesterday as well. Maybe all this stay-at-home-because-covid19-stuff is messing with my head...

If anyone else experience a similar problem, then try this:
```bash
ldd /bin/* | grep curses
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f8951d98000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007ff92a888000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f557521d000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007fda2eff6000)
	libncurses.so.6 => /lib/x86_64-linux-gnu/libncurses.so.6 (0x00007f0f88fa9000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007efefc9aa000)
/bin/ncurses5-config:
/bin/ncurses6-config:
/bin/ncursesw5-config:
/bin/ncursesw6-config:
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f376cc6d000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f73376f9000)
/bin/pinentry-curses:
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f7c00bf7000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007fa7d676e000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f8e0038d000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f4291b96000)
	libncurses.so.6 => /lib/x86_64-linux-gnu/libncurses.so.6 (0x00007fc7f9745000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007fc57c61e000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007f3c1fa0f000)
```

In my case only libncurses6-files showed up. To install version 5:

```bash
sudo apt-get install libncurses5
```
Note that the package is libncurses5 and not libncurses5-dev.

@haneefdm thanks for great assistance and response time! :D

closed 2020-03-26 by aakre

haneefdm, 2020-03-26:
Sorry for me being testy :-) It rarely happens, but it does.

Yes, this was reported once before but I didn't think of it. Thanks for posting the solutions.

pearson, 2021-06-10:
@aakre said:
> sudo apt-get install libncurses5

THANK YOU!  I ran into the same issue and this solved it.

gergamel, 2022-06-09:
Just some additional information to help anyone else who has a similar setup. I hit this issue on a Fedora machine and the package you need to install is ```ncurses-compat-libs```.

My setup was:

- Fedora 36 (5.17.12-300.fc36.x86_64)
- gcc-arm-none-eabi-10-2020-q4-major
- SEGGER J-Link GDB Server V7.66b