Marus/cortex-debug #268: Project page appears to be down

cdwilson, 2020-03-30:
I'm getting this error when trying to visit https://marcelball.ca/projects/cortex-debug/

[image]

haneefdm, 2020-03-30:
Isn't the real link https://github.com/Marus/cortex-debug

It is hosted on github.

cdwilson, 2020-03-30:
Hi @haneefdm I'm new to the cortex-debug project and I'm trying to find more documentation on the available configuration options.

I just clicked the link on the GitHub page looking for documentation:

[image]

[image]

Can you send me the updated link to the documentation hosted on GitHub?

haneefdm, 2020-03-30:
Oh sorry, that link is a private link owned the author of this extension @Marus It appears his website is down

I will try to contact him. The last time I checked, that documentation is pretty old and was not being maintained.

I always used Intellisense in VSCode to see what all is available. I will see how this can be addressed.

cdwilson, 2020-03-30:
Thanks for the tip on Intellisense, it gave me enough hints to get things working.  I do think having updated documentation would make the setup process smoother for new users.  I just realized that there is some very basic documentation on the github wiki at https://github.com/Marus/cortex-debug/wiki  Are you guys planning to use that moving forward?

haneefdm, 2020-03-30:
I wish I could use 'https://marcelball.ca/projects/cortex-debug/' as a starting point.

Also, looking for volunteers I am not a good technical writer. I will give it a shot when I have some time.

haneefdm, 2020-03-30:
TBH, I never knew this wiki existed :-) I added a whole bunch of missing properties for launch.json

Didn't need a whole bunch of technical writing skills!!

cdwilson, 2020-03-30:
Seems like having this info on the Wiki would allow it to stay up to date more easily.  The wayback machine cache of this project page has some nice detailed documentation, but it looks outdated... maybe that info can be ported over to the wiki if @Marus doesn't want the hassle of keeping the personal site maintained?

https://web.archive.org/web/20190504094943/https://marcelball.ca/projects/cortex-debug/

haneefdm, 2020-05-14:
We will be moving the documentation to the Wiki and the project page will be updated to remove the obsolete link

https://github.com/Marus/cortex-debug/wiki

closed 2020-05-14 by haneefdm