Marus/cortex-debug #270: J-Link debugging no longer working

barak14, 2020-04-01:
Hi there, 
2 days ago everything was working fine, then i think the microsoft c++ extension updated and since then i cannot get cortex-debug to work.
The debug console does not update when i start debugging and there is no entry under the output->remote adapter that i used to be able to see when there was a problem with segger.
Furthermore I have code insiders installed separately and when the issue presented itself in this version of code I was still able to debug using the insider version with the same workspace and output file. Then the cpp plugin updated and it all stopped in my insiders copy too. 
I have removed the cpp extension and re-added it but that has not resolved the issue (same with cortex-debug)

code 1.43.2

my launch is like this
{
      "name": "App",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}\\app\\${config:iarvsc.configuration}\\Exe\\${config:iarvsc.configuration}.out",
      "request": "launch",
      "type": "cortex-debug",
      "svdFile": "${workspaceRoot}/STM32F4/STM32F40x.svd",
      "servertype": "jlink",
      "device": "STM32F407VE",
      "interface": "jtag",
      "showDevDebugOutput": true,
      "runToMain": true
    },
Edit: I have just enabled showDevDebugOutput i can see the following in the debug adapter:
000000016: Reading symbols from 'C:\xx\yyy\(folder with brackets)\Exe\(filename with spaces and brackets).out' but it just sits there and doesn't proceeed.

Any ideas?
Thanks for you efforts!

haneefdm, 2020-04-01:
Is it possible for me to get the executable? That part did change recently. I don't need the source.

haneefdm, 2020-04-01:
It will also help to know the version of the arm toolchain. If you can't share the executable, can you do the following for me
```bash
arm-none-eabi-objdump -Wi --syms elffile.out > elffile.dump
```
Also, bit more info. Do an `About VSCode`, please copy-paste the details. I am also using code 1.43.2 on a Mac and Windows and things are working fine.

barak14, 2020-04-01:
Sorry only saw this in the morning - I managed to recreate this with a much smaller binary that i don't mind sharing, basically just turns on the display on my unit and that's it.

Hmm i am using IAR v8.50 but it occurs with all versions that have been working 7.70 and 8.40 too.
[test.zip](https://github.com/Marus/cortex-debug/files/4418136/test.zip)

Version: 1.43.2 (user setup)
Commit: 0ba0ca52957102ca3527cf479571617f0de6ed50
Date: 2020-03-24T07:38:38.248Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.19592

haneefdm, 2020-04-02:
You can use IAR/Keil for compile/link but we still have to use ARM GNU tools gdb/objdump for debugging.

What does `arm-none-eabi-objdump --version` print?

barak14, 2020-04-02:
Sorry! I have tried 3 different versions of arm-none-eabi-gdb that i have installed:

`GNU objdump (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 2.28.0.20170620`

`GNU objdump (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 2.32.0.20190703`

`GNU objdump (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 2.33.1.20191025`

haneefdm, 2020-04-02:
The elf file format does not match what we expect out of gcc for symbols. May I ask the command line used to compile with IAR -- the exact command-line to use a sample C file? Were full path-names used to compile each file?

haneefdm, 2020-04-02:
Oh, we got a BIG problem. I think we found a bug in Node.js. We try to match a regexp for output of objdump and it never returns on the very first call. Supposed to either match or fail but it never returns.

No idea how to resolve this. Perhaps I can find a workaround. You can revert to an earlier version of Cortex-Debug until I can figure it out.

Please be patient. VSCode, Node.js, and Cortex-Debug changed in a matter of days.

barak14, 2020-04-02:
I'm just glad it's not only me! 😏 
Thanks for looking into this - I can still use the IAR IDE to perform debugging no matter how clunky that can feel after using vscode for a while.
How do I go about reverting the version of Cortex-Debug?

microwavesafe, 2020-04-03:
I'm seeing similar problems and I don't use the Microsoft C++ extension (I use CCLS instead). I thought that might be worth adding as it doesn't appear to be related to that extension.

I can confirm that initial results suggest reverting to 0.3.5 fixes my issues.

mischelebuha, 2020-04-03:
We have almost the same problem. But only for the project with an IAR extension. When starting the debugging process, nothing happens for a while (very long). After that, debugging works without problems. And the problem is solved by using 0.3.5.

In another project with the Microsoft C ++ extension everything works.

haneefdm, 2020-04-03:
This is specific to IAR produced elf files and this version of Node.js. And yes, please use 0.3.5 until I figure out a  fix.

microwavesafe, 2020-04-03:
The bugs I was seeing with 0.3.6 was the variables box was empty, no local variables were shown at all when stepping through code. And occasionally the watched variables would stop updating, with a rotation animation near them. To restore the watch variables I could reconnect the debugger, but the variables window never came back.

This is with files produced with the GCC toolchain. Is this a different bug? Should I start a new issue?

haneefdm, 2020-04-03:
@microwavesafe, I think this a different problem. Could you start a new issue and detailed steps to reproduce?

haneefdm, 2020-04-03:
@barak14 I found a new problem. In the symbol file, there is some information that is not conformant to the documented output of objdump and these symbols are being missed.
```
00000000 l    df *ABS*	00000000 C:\Development\GitRepos\Firmware\phoenix\library\ap_config.c
08004edc l     O P2 ro	0000006c s_inbuilt_mmap
20000000 l     O P3 rw	00000001 s_model_type
20000004 l     O P3 rw	00000004 s_serial
0800c000 l     O P1	00000008 s_serial_number
```
The fields/letters `rw`, `ro` after P2/P3 are not expected. In the above file, `s_serial_number` is recognized as a valid symbol but not the other three. Could you tell me what these variables are? I can guess `ro=read-only`, `rw=read-write` but what do their C declarations look like?

While it looked like it was working before just a few symbols matched the expected format. Rest were ignored

pfox89, 2020-04-03:
@haneefdm Another IAR user here. This fix seems to resolve this issue for me. Thanks!

pfox89, 2020-04-03:
@haneefdm Regarding the extra field in objdump, I'm seeing 4 different designations in my code: ro, s0, zi, and ui.

- ro looks like text and rodata (functions and const declarations).
- s0 looks like initialized variables (static initialized?)
- zi looks like zero-initialized (e.g. static variables without assignment)
-  ui appears to be uninitiated variables (e.g. stack, heap).

It's odd that objdump supports these designations, but doesn't seem to have them documented...

pfox89, 2020-04-03:
@haneefdm Actually, looking at it a bit further, these designations actually appear to be part of the section name. Apparently you can have spaces in a section name? Interesting.

haneefdm, 2020-04-03:
Yeah, traditionally, they are supposed to be things like .bss, .rodata., .text, etc. in the previous field. I did not know you can have spaces in a section name.

@pfox89 Thanks. Do you have a reference to where it says that a section name can have white space in it?

pfox89, 2020-04-03:
@haneefdm Well, interestingly I can't find anything that specifically says you *can* do it, but there also don't seem to be any rules *against* it. If you look at this document: https://refspecs.linuxfoundation.org/elf/gabi4+/ch4.strtab.html
basically anything except a 0 byte appears to be an allowed character in a section name.

It might not be a good *idea* to have spaces in a section name, but that doesn't seem to have stopped IAR. In fact, I can use ```objdump -j "P2 ui" -t HelloWorld.elf``` and get a list of symbols (once I know what section names exist).

I think the reason they do this might have to do with the way the IAR linker does placement. It doesn't actually create ELF sections for the internal sections you define. Instead, it creates a section for each memory region you defined and places your sections according to how they're categorized. Then I suppose it must break them down by what kind of initialization is needed.

haneefdm, 2020-04-03:
@pfox89 Thanks. Right now with 0.3.7-pre2, it works if there are two words in the section name separated by white space. I may have to adjust the parser to arbitrary stuff and make another release.

barak14, 2020-04-15:
Thanks for the 0.3.7 version - it has fixed my issue!

closed 2020-04-29 by barak14

TorBlackbeard, 2020-05-07:
Seems to work for me!