Marus/cortex-debug #272: Missing local variables

microwavesafe, 2020-04-03:
Using 0.3.6 occasionally the local variables disappear, the window is there only empty.

[image]

I can disconnect and reconnect to the debugger, and they do not return. But if I restart VSCode they usually turn back up. Reverting to 0.3.5 seems to fix the issue.

It is difficult to reproduce as it just seems to happen. I have captured three debug logs. The first is long as this is when it starts off working, then at the end they disappear. The second log is restarting the debug session with the local variables still empty. The third is the log from a newly started window with the local variables back again.

I am using the GCC toolchain (I have tried two from 2017 and 2019 with the same results).

[first_log.txt](https://github.com/Marus/cortex-debug/files/4427635/first_log.txt)
[second_log.txt](https://github.com/Marus/cortex-debug/files/4427637/second_log.txt)
[third_log.txt](https://github.com/Marus/cortex-debug/files/4427638/third_log.txt)

haneefdm, 2020-04-03:
How are you restarting the session? Using the Restart button or end session and then launch again?

haneefdm, 2020-04-03:
Also, when you see the variable window empty, does it fix itself if you single step?

microwavesafe, 2020-04-03:
I am restarting the debug session using the stop button of the debug window (with the go / pause, single step). Then hitting F5 to restart it.

No, single stepping does not fix it. The window remains empty whatever I try until I restart VSCode completely.

haneefdm, 2020-04-03:
Oh, this is with an attach vs. a launch?

haneefdm, 2020-04-03:
Btw, what does your `CALL STACK` window look like? Empty. Then we have to focus there. If there is not stack, there will be no variables either.

haneefdm, 2020-04-04:
@microwavesafe Could you try out the pre-release mentioned above. please and report back. I cannot reproduce what you are seeing.

microwavesafe, 2020-04-04:
I will try this out as soon as possible and get back to you.

microwavesafe, 2020-04-06:
The pre-release seems to have fixed the issue. I've been testing for 30 minutes and it seems much more stable. It also seems to have cleaned up a few other issues. I was getting a number of error boxes pop up about not retrieving variables when pausing execution. I was also getting "unknown" as a value in the variable window sometimes, even though adding the same variable to the watch window worked fined.

If it happens again I will check the call stack as requested before.

closed 2020-04-12 by microwavesafe