Marus/cortex-debug #273: Parsing issue of SVD file

josko7452, 2020-04-14:
Hi I have trouble getting SVD file loaded from:
https://github.com/m-labs/dslite2svd/blob/master/svd/tm4c123x.xml

I think it will be likely related to #232 or it is a different parsing error.

haneefdm, 2020-04-15:
Could you post the error(s) you are seeing?

josko7452, 2020-04-15:
It's just No SVD File Loaded <PATH>
Where <PATH> is correct path to my file (i.e. E:\WORK\Data\tm4c123x.svd)

haneefdm, 2020-06-08:
Found the issue, fix is coming. We were not expecting peripherals without an `addressBlock`, and "_INTERRUPTS" was one such peripheral.

haneefdm, 2020-06-13:
Could you please try the following pre-release

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre3

Use the VSIX manual installation method to install the extension

josko7452, 2020-06-15:
@haneefdm It works on the file I've linked above.

closed 2020-06-15 by josko7452