Marus/cortex-debug #275: file path in debug console not right. 'cwd' not help.

qingyecai, 2020-04-17:
Hello everyone , in my setting
[image]
When i start debug, in debug consle print wrong file path.
[image]
This is my project path:D:\aCorporation\Commnet\gitlab\SmartGateway
This is my elf path:D:\aCorporation\Commnet\gitlab\SmartGateway\build\stm32\SmartGateway.elf
I try to set the vaule 'cwd' in lauch.json, but it can't help.
Can anybody help me fix it out?
Thinks.
Best Regards.

haneefdm, 2020-04-17:
I don't think I understand the problem. gdb displays path names that are relative to how the source files are compiled. Ideally. It will resolve them if it can relative to the 'cwd' but if they are different from compilation directory vs workspace dir, then you may have problems. I have a feeling, your `cwd` should be set to `D:\aCorporation\Commnet\gitlab\SmartGateway` and not where the elf file lives.

Also, when you click on the file name, that is handled by VSCode and not this extension.

qingyecai, 2020-04-20:
@haneefdm 
I have try  ```cwd:D:\aCorporation\Commnet\gitlab\SmartGateway``` but it's not help .
You mean I shoud  check my ```Cmakelist.txt```? or the ```Build``` task?
Thank you for your help.
Best Regards.

qingyecai, 2020-04-20:
I have change ```"command": "cmake -G Ninja \"${workspaceFolder}\" -DCMAKE_BUILD_TYPE=Debug" ``` to ```"command": "cmake -G \"MSYS Makefiles\" \"${workspaceFolder}\" -DCMAKE_BUILD_TYPE=Debug",```
everything is ok!

closed 2020-04-20 by qingyecai