Marus/cortex-debug #278: Using cortex debug from wsl

trondhe, 2020-04-24:
Any good pointers on how using cortex debug from wsl should be set up?

Due to the setup im limited to, I've come to the point where using vscode via wsl is the best solution and works great for programming/building/git. But I've hit a snag when it comes to debugging, and cant get anything to work.

Its a nrf52840 devboard that is used right now.
My idea was to start JlinkGdbServer from windows, and then connect to it from wsl via localhost.
The jlinkgdbserver command
```
JLinkGDBServerCL.exe -device nrf52840_XXAA -endian little -if SWD -speed 4000 -rtos GDBServer/RTOSPlugin_FreeRTOS
```
and the launch.json
```
{
    "name": "debug",
    "type": "cortex-debug",
    "request": "attach",
    "servertype": "jlink",
    "showDevDebugOutput": true,
    "debugServer": 2331,
    "executable": "${workspaceRoot}/_output/myoutput.elf",
    "device": "NRF52840_XXAA",
    "interface": "swd"
 },
```
The gdbserver will be listening on port 2331, and when I press debug it says
`Connected to 127.0.0.1` but no further output is given anywhere, neither in the gdbserver, vscode adapter output, debug console or terminal.
Anything obvious I'm overlooking?

haneefdm, 2020-04-24:
Do not use "debugServer". Use "gdbTarget" to specify "host:port" and "servertype" should be "external" if you are starting the JLinkGDBServerCL by yourself.

Also, things like "interface", "device" do not apply.

"request" can be "attach" if you are manually reset the board/device or "launch" if you want Cortex-Debug to do a reset for you.

There will be nothing in the "Adapter output" window since you own the adapter now. But, debug console will be normal looking.

There are people using this with WSL.

https://github.com/Marus/cortex-debug/issues?q=is%3Aissue+WSL

trondhe, 2020-04-24:
Thanks so much for the quick reply!
Its a bit of a mouthful all these layers, hard to keep track, I'll check it out.

closed 2020-04-24 by trondhe