Marus/cortex-debug #279: FreeRTOS Debugging Issue on JLink GDB Server

gadak, 2020-04-27:
Hey!

I am using following - 
CPU-IMX7D
OS-FreeRTOS

When I keep a breakpoint in my task or even if I don't keep one, the debug status doesn't change after pressing run. It always remains in paused state and '[Switching to Thread 539100480]' is printed in debug output. 
Projects without RTOS are debugging properly. but have issue with RTOS. Any thing that I can do to get over this issue? 

 with following launch.json
`{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "J-Link Debug rpmsg_gpt_freertos.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "MCIMX7D5_M4",
            "cwd": "${workspaceRoot}",
            "executable": "Debug/rpmsg_gpt_freertos.elf",
            "runToMain": true,
//            "rtos": "/opt/SEGGER/JLink_V670g/GDBServer/RTOSPlugin_FreeRTOS.so",
            "rtos": "FreeRTOS",
            "postLaunchCommands": [
                "flush regs",
                "load"
            ],
            "postRestartCommands": [
                "flush regs",
                "load"
            ],
            "armToolchainPath": "/home/kashyap/tdx-freertos/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux/gcc-arm-none-eabi-9-2019-q4-major/bin",
            "gdbTarget": "/opt/SEGGER/JLink_V670g/JLinkGDBServer"
        }
    ]
}`
Output- 
`SEGGER J-Link GDB Server V6.70g Command Line Version
JLinkARM.dll V6.70g (DLL compiled Apr 23 2020 09:14:02)

Command line: -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MCIMX7D5_M4 -rtos GDBServer/RTOSPlugin_FreeRTOS
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 MCIMX7D5_M4
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Hardware: V7.00
Feature(s): RDI,FlashDL,FlashBP,JFlash,GDBFull
Checking target voltage...
Target voltage: 3.30 V
[... 117 lines cut ...]
Removing breakpoint @ address 0x0091408A, Size = 2
Gtk-Message: 21:14:47.526: Failed to load module "canberra-gtk-module"
`
Debug Console Output- 
`Please check OUTPUT tab (Adapter Output) for output from JLinkGDBServer
Launching server: "JLinkGDBServer" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "MCIMX7D5_M4" "-rtos" "GDBServer/RTOSPlugin_FreeRTOS"
Launching GDB: "/home/kashyap/tdx-freertos/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /home/kashyap/tdx-freertos/eclipse-workspace/rpmsg_gpt_freertos/Debug/rpmsg_gpt_freertos.elf...
0x00910622 in prvPortStartFirstTask () at ../sources/rtos/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.c:290
290		__asm volatile(
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
Register cache flushed.
Loading section .interrupts, size 0x240 lma 0x0
Loading section .text, size 0x5d68 lma 0x910000
Loading section .ARM, size 0x8 lma 0x915d68
Loading section .init_array, size 0x4 lma 0x915d70
Loading section .fini_array, size 0x4 lma 0x915d74
Loading section .data, size 0x114 lma 0x20220000
Start address 0x9100b4, load size 24780
Transfer rate: 84 KB/sec, 3540 bytes/write.
[New Remote target]
[New Remote target]
[Switching to Thread 539100480]`

haneefdm, 2020-04-27:
Does doing a continue followed by a pause work?

I never used `"postLaunchCommands"` like that. Why was that needed? Especially the `load`. Wouldn't that re-load the executable and destroy your reset state and the current breakpoint? A load was already done for you following a reset.

haneefdm, 2020-04-27:
Also isn't `flushregs` one word rather than `flush regs`. And in `postRestartCommands` doing a 'load' is not expected and not sure what that would do.

haneefdm, 2020-04-27:
When you do a load, normally pretty much all your execution context is lost because JLink or OpenOCD has to load a temporary program into SRAM to do the programming of the device, so you are destroying all your currents RTOS-threads/tasks/PC/stack/etc.

gadak, 2020-04-27:
> Does doing a continue followed by a pause work?
No
> I never used `"postLaunchCommands"` like that. Why was that needed? Especially the `load`. Wouldn't that re-load the executable and destroy your reset state and the current breakpoint? A load was already done for you following a reset.
I actually don't know what it does and it was there in the default script. So I didn't make any changes.

haneefdm, 2020-06-14:
Please point me to a real issue. I really don't understand or help me duplicate it.

You have to understand what is in your launch.json. Please talk to whoever gave you that file. I am closing the issue for now.

closed 2020-06-14 by haneefdm