Marus/cortex-debug #280: Can't find openocd.cfg

ksv767, 2020-04-27:
I am trying to set up the launch.json file to run the debugger.
I always get the following error message even when I give the entire path.

embedded:startup.tcl:60: Error: Can't find  openocd.cfg

???

haneefdm, 2020-04-27:
Well, it would help to attach the launch.json? Also, for any given device/board you have to at least one configuration files using the `configFiles` setting if you are using OpenOCD. Without that OpenOCD does not work.

ksv767, 2020-04-27:
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "gnu-debugger",
            "request": "launch",
            "name": "GNU debugger",
            "program": "${workspaceFolder}/build/VSM3-CB.elf",
            "toolchain": "/usr/bin",
            "client": "arm-none-eabi-gdb",
            "clientArgs": [],
            "server": "openocd",
            "serverArgs": ["-f ./openocd/openocd.cfg"],
            "linux": {
                "MIMode": "gdb",
                "miDebuggerPath": "/usr/bin",
                "server": "/usr/local/openocd/bin/openocd",
                "serverPort": 3333
            },
        }
    ]
}

configFiles setting?

ksv767, 2020-04-27:
openocd works fine from my command line, I just cannot get the launch.json file set up correctly so that it works from Visual Studio Code with the debugger plugin

haneefdm, 2020-04-27:
No configFiles is a separate setting in launch.json. Please use that instead of serverArgs. Sure you may be able to use server args but even so, you combined two args into one string. Not how it works even from command-line. You can clearly see the command-line used in the Debug Console based on your launch.json

haneefdm, 2020-04-27:
I am also not sure where you got the rest of the launch.json settings. None of it makes sense. You may have taken some other debuggers launch settings and just changed the debugger. Looks totally random stuff to me.

Please use VSCode to generate a template and then add to it. If you open a launch.json, you will see a big button on the bottom right 

[image]

Use Intellisense while editing launch.json and you can find valid settings and what they mean.

closed 2020-04-27 by haneefdm