Marus/cortex-debug #282: gdb launches, openocd launches, never reaches main

ksv767, 2020-04-27:
With the launch.json below, gdb and openocd launch, however the application never reaches main.

Is there more that needs to be added to the launch.json?

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [

        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/VSM3-CB.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "serverpath": "/usr/local/openocd/bin/openocd",
            "configFiles": ["./openocd/openocd.cfg"],
        },
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/VSM3-CB.elf",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "setupCommands": 
            [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                },
                {"text": "target remote : 3333"}
            ]
        },
    ]
}

haneefdm, 2020-04-27:
Well, you didn't use `runToMain` or set a breakpoint?. Please use IntelliSense to discover all the options.

ksv767, 2020-04-28:
Updated launch.json - and still doesn't reach main - please see output console after launch.json listing below:

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [

        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/VSM3-CB.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "device":"STM32L4A6xx",
            "servertype": "openocd",
            "serverpath": "/usr/local/openocd/bin/openocd",
            "configFiles": ["./openocd/openocd.cfg"],
            "rtos": "FreeRTOS",
            "runToMain": true,
        },
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/VSM3-CB.elf",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "miDebuggerArgs": "target remote : 3333",
            "setupCommands": 
            [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                },
            ]
        },
    ]
}

Please check OUTPUT tab (Adapter Output) for output from /usr/local/openocd/bin/openocd
Launching server: "/usr/local/openocd/bin/openocd" "-c" "gdb_port 50000" "-s" "/home/ken/Projects/maven/comms_board/pui_new" "-f" "./openocd/openocd.cfg" "-f" "/tmp/tmp-20590gzF4c7uGJIYn"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from /home/ken/Projects/maven/comms_board/pui_new/build/VSM3-CB.elf...
**0x00000000 in ?? ()
Not implemented stop reason (assuming exception): undefined**
Unable to match requested speed 500 kHz, using 480 kHz
Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800d420 msp: 0x20050000

haneefdm, 2020-04-28:
> Please check OUTPUT tab (Adapter Output) for output from /usr/local/openocd/bin/openocd

You have to see what openocd is doing. You can use the exact same command lines as what Cortex-Debug uses from a couple of terminals. I am not sure your `configFiles` is complete. If you have another debugger that works, then you can copy their configuration here. Or contact the vendor. With OpenOCD, you have to understand how it works especially for configuration files

In launch.json you can enable `showDevDebugOutput` to see all the commands sent to gdb and its responses. If you cannot debug via command-line, you cannot debug via Cortex-Debug.

Your launch.json also has Microsofts `cppdbg` debugger. Did that work? Probably not.

haneefdm, 2020-06-08:
I see no response from the submitter

closed 2020-06-08 by haneefdm