Marus/cortex-debug #283: graphConfig

marvelou-s, 2020-04-30:
The goal is to get separate runtime graphs during debug. How could I attach data from different ITM channels? I found the only one last year example (#139), but it's not clear and maybe not actual already. I didn't find any documentation in the same time.

Looking for any suggestions. Thank you!

PhilippHaefele, 2021-12-23:
@marvelou-s I think https://github.com/Marus/cortex-debug/wiki/SWO-Output#vscode-cortex-debug-configuration documents what you wanted to do. The only thing to mention is that port = ITMChannel.

To have multiple graphs just add another entry to the "graphConfig" array (not tested, but as it is an array it‘s the obvious thing to go).

If you have any further questions just state that here so we can extend the documentation.

@haneefdm Please close the issue. Thanks

haneefdm, 2021-12-23:
Thanks @PhilippHaefele

closed 2021-12-23 by haneefdm