Marus/cortex-debug #284: SVD File not loading the derivedFrom values

danielhep, 2020-04-30:
My SVD file makes heavy use of the "derivedFrom" attribute on some registers, since many registers share the same set of fields. However, in cortex-debug, the fields only show up on the registers on which they are defined, and not those which are derived from them. 

Here's an example of the register that uses derivedFrom:
```
                <register derivedFrom="P300PFS">
                    <dim>10</dim>
                    <dimIncrement>0x4</dimIncrement>
                    <dimIndex>0-9</dimIndex>
                    <name>P00%sPFS</name>
                    <description>P00%s Pin Function Control Register</description>
                    <addressOffset>0x000</addressOffset>
                    <size>32</size>
                    <access>read-write</access>
                    <resetValue>0x00000C00</resetValue>
                    <resetMask>0xFFFFFFFD</resetMask>
                </register>
```
The P300PFS register has a <fields> tag on it that I would expect to carry forward to the registers that result from the above, however as you can see in this screenshot, only the "parent" registers have any fields on them. 

[image]

Based on what I read [here](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_register), it seems like this syntax is correct, since both registers are located in the same peripheral, so it should not be necessary to use the peripheral.register notation. If this is a problem in cortex-debug, feel free to point me in the right direction for fixing it, and I can submit a pull request.

haneefdm, 2020-06-08:
Is it possible to have the full svd file?

haneefdm, 2020-06-11:
Can you provide more info, please? Debugging SVD file issues is not easy. Please respond @danielhep I need the actual SVD file.

I am about to make a new release and hope to get your issue resolved before then. I am sure you have a real issue but I need to be able to reproduce it.

danielhep, 2020-06-12:
@haneefdm I sent you the SVD via email. Let me know if that works for you. Thank you!

haneefdm, 2020-06-13:
Could you please try the following pre-release

https://github.com/haneefdm/cortex-debug/releases/tag/v0.3.7-pre3

Use the VSIX manual installation method to install the extension

danielhep, 2020-06-15:
It looks like the registers are showing up, but they are not showing accurate information. Also, writing to them doesn't seem to work. These registers control an IO connected to an LED, so that is what I'm testing with. 

In this screenshot you can see the registers are reported correctly in the upper "Watch" pane, but it shows as all zeros in the bottom "Peripherals" pane. 

https://imgur.com/a/FQ3OtnW

haneefdm, 2020-06-15:
Thanks @danielhep for getting back to me. Looks like while the fields are being recognized, the address seems to be not right

I have to conjure some example (map the base address to SRAM) to figure out what is going on.

haneefdm, 2020-06-20:
I believe your issue has been addressed in 0.3.7. Thanks for your help

Please reopen the issue if it still a problem

closed 2020-06-20 by haneefdm

danielhep, 2020-06-21:
Thank you kindly! I will give it a try on Monday and let you know if it's still a problem.