Marus/cortex-debug #285: ST-LINK-SERVER Support?

tdjastrzebski, 2020-05-01:
Any plans to support ST-LINK-SERVER?
https://www.st.com/en/development-tools/st-link-server.html  
https://www.st.com/resource/en/release_note/dm00402226-stlink-server-stmicroelectronics.pdf
It seems it is a relatively new utility, actively supported on Windows, macOS X, Linux Debian, Linux Red Hat.

haneefdm, 2020-05-01:
I just downloaded and installed it on my Mac. I have no idea what it did, where it installed, what the executable name is :-)

If it is like any other gdb server, it is not that difficult to add. I have to first check with the owner of this extension to see if it is okay to add.

But, please educate me a bit more.

Marus, 2020-05-01:
Yeah, now that they have an official one I think it would be a good idea to
support it. Will have a look - but as Haneef said should be
fairly straightforward to add the support.

sidprice, 2020-06-18:
@Marus I would really like to see this, I love Cortex-Debug, however I want to use ST GDB Server please. The "official" ST GDB Server has been available for a long time now.

haneefdm, 2020-06-19:
@sidprice Have you tried the gdb-server `"external"` type? You can use that and connect to any server (local or remote). Use `"gdbTarget"` to specify host:port

I mean try it for now. As I said earlier, when I installed it, I have no idea what it actually did, how to launch it or use it. I asked for help earlier as well.

sidprice, 2020-06-19:
@haneefdm Thank you, that does not work, it gets a timeout error. 

@Marus It would be great to support the ST server, so many Nucleo/Discovery boards that could be interfaced with, without openOCD.

RisinT96, 2020-06-19:
@sidprice hey, I might have a temporary solution for the on-board st link debugging, you can turn it into an on-board j-link, which works great with this plug-in.
More info here:
https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/

Imo it works even better than the original on-board stlink.

sidprice, 2020-06-19:
@RisinT96 Thank you, I know about that use it often. This is a situation where I MUST use STLink and not JLink.

haneefdm, 2020-06-19:
@sidprice Guess you decided not to help me fix/add support for the ST Server.

`"external"` means you have to start the gdb-server/ST-server manually yourself and then start deebugging. If YOU can't start the server manually, neither can this extension.

I CAN really make the support happen. I am the only other maintainer of this extension.

I am going to remove myself from this thread. Enjoy.

RisinT96, 2020-06-19:
> @RisinT96 Thank you, I know about that use it often. This is a situation where I MUST use STLink and not JLink.

Well, @Marus seems to have disappeared 3 months ago with no activity since, and @haneefdm  is the only one trying to maintain this extension.

haneefdm, 2020-06-19:
@Marus is a bit busy, he lurks, he will be back. I am trying to do the best I can, with @Marus permission.

We do have day jobs :-)

sidprice, 2020-06-19:
The server was started with a known port number and it did not work. I understand having a day job, I have one too. If I was a typescript person I would help, Thanks.

haneefdm, 2020-06-19:
@sidprice You are not giving me any details what so ever. No logs, screenshots, no pointers to how I can use the ST server. Nothing. You just complain.

You have a day job and I bet you are using this tool in your day job. But unwilling to help me help up. Talk about entitlement. You are as disrespectful as one can be. I am sure you can do worse though.

Bye Sid.

sidprice, 2020-06-19:
@haneefdm Well what an attitude, that is sure to get the project more help. I stated all the information I have, I see a timeout error. Perhaps if you let me know what additional information you need or what testing I can do I could help,

sidprice, 2020-06-19:
@haneefdm Here is some deeper information for you, it seems the issue is with the current version of the ST GDB Server that is a part of the STM32CubeIDE, starting it is a little more complex that the older versions. I will try to find out what is needed and report it back here so that, if I canmake it work, others may use it too.

tdjastrzebski, 2020-08-01:
@sidprice @haneefdm @RisinT96 Is there any hope or this St-Link v3 is just too difficult to work with? - which would not surprise me.

haneefdm, 2020-08-01:
@tdjastrzebski I recently ordered a board with ST-Link V3 with ST's gdb-server. No promises but maybe I can look at it next weekend.

But in the meantime, people have been using V3 with OpenOCD. See also Issue #228

tdjastrzebski, 2020-08-02:
@Marus @haneefdm @sidprice @RisinT96 Although ST-Link-Server details may not be publicly available, I think it may be worthwhile to contact ST. After all, they built this software to increase their hardware sales. Depending on the level of their internal politics, it may be possible to get some help from them.   
Cortex-debug is a recognized solution, maintain by engineers working together not to make a profit but to make their own and others life easier.  
I would gladly help with this effort, initially probably by testing.  
Yesterday I managed to connect ST-Link Mini V3 to Nucleo-F746ZG board. I am able to flash it and communicate over serial interface but no luck with debugging yet.  
I think this probe is very promising: both SWO and JTAG support, Serial, I2C, SPI, Can + 4 GPIOs in one small, inexpensive package. I think supporting this probe makes a lot of sense, providing its software is good enough - which I do not know yet.
[image]

JojoS62, 2020-10-16:
I have used a STLink V3 on a Nucleo H7A3 board via pyocd in cortex-debug. I could use the debugger for single stepping, but flashing the MCU didn't work. I have entered a 'load' command manually again, that showed an amazing fast programming speed, but the existing program was not changed.
Are there any updates on using STLink V3?

Marus, 2021-02-14:
The official ST-Link GDB server is now supported on Windows. We’ve been having issues with getting it to launch properly with from VSCode on Linux and macOS as for some reason it cannot find its custom shared libraries needed. 

I’m going to close this particular ticket as there is another one open.

closed 2021-02-14 by Marus