Marus/cortex-debug #286: download works ... then later doesn't ... then does again

TorBlackbeard, 2020-05-06:
It's driving me nuts, first download+debug works fine ... then it doesn't ... later it does again.
Symptom is that I get only one line of debug-output:
0000000010: Reading symbols from 'xxx\debug\release\project.out'
and nothing else happens. 

```
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "build/efm32frontend/debug/release/project.out",
            "name": "Run ...",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "EFM32GG990F1024",
            "svdFile": "./EFM32GG9xx.svd",
            "runToMain": true,
            "armToolchainPath": "C:\\Program Files (x86)\\GNU Tools Arm Embedded\\9-2019-q4-major\\bin",
            "serverArgs": [
                "-select USB",
                "-if SWD"
            ],
            "showDevDebugOutput": true
        }
    ]
```

When problem is there, I can always start the Jlink server this way:
            JLinkGDBServerCl.exe -select USB -if SWD -device EFM32GG990F1024
And connect to it with 
  arm-eabi-none-gdb file.out
And connect to it with "target remote :2331" and "load".

Tried investigating:
- if the vscode and jlinkgdbservercl was firewalled by my McAffee suite (company policy ...)
  Aparantly not, since I can do it manually on the command line as described
- if the ELF file for some reason has offending contents ...
  I'm not completely sure ... The ELF was build by IAR compiler. But as described, it can at least be loaded by arm-eabi-none-gdb successfully. If it is offending to cortex-debug, I don't know.

Tried installing it on a co-workers computer, where it started up in bad-mode. An within the 5 minutes we gave it, we didn't succeed making it work.

Am I missing something obvious here?
:-)

haneefdm, 2020-05-06:
Perhaps, this related to issue #270 

There is a partial fix available. See

https://github.com/Marus/cortex-debug/issues/270#issuecomment-608611176

Still working on final fix.

TorBlackbeard, 2020-05-07:
Works ... :-)
I'm still baffled about why the ELF is *sometimes* offending... but it works!

By the way:
Thx to the entire cortex-debug team. 
VsCode + this plugin saves me from even installing Eclipse on my computer ... you can't imagine the joy that brings me!

closed 2020-05-07 by TorBlackbeard

haneefdm, 2020-05-07:
IAR produces elf files that have unexpected section names (multi-word, non-standard section names). Technically, it is a legal elf file. Turns out it was actually a bug in Node.js that was causing the hang.

And thank you for appreciating our work. It is just two guys. @Marus is the main person.