Marus/cortex-debug #287: Add option to specify custom openocd.gdb conf file

andresv, 2020-05-07:
It would be useful for example to enable semihosting in `openocd.gdb`:
```
target remote :3333
set print asm-demangle on
break DefaultHandler
break HardFault
break rust_begin_unwind
monitor arm semihosting enable
load
break main
continue

```

Related: https://github.com/rust-embedded/cortex-m-quickstart/issues/85

haneefdm, 2020-05-07:
It sounds like you don't want to use this extension. It already does all of that automatically

And, there are tons of ways to customize it. Play with launch.json and use IntelliSense and there are a host of options customize. Even the default configuration you create in VSCode will probably suffice.

Setting your own breakpoints is not a good idea in VSCode because it manages all of them. Not even this extension.

closed 2020-05-07 by haneefdm

andresv, 2020-05-07:
I found this: https://github.com/haneefdm/cortex-debug-samples/blob/master/blink2/.vscode/launch.json

And added those lines for enabling semihosting:
```
...
"postRestartCommands": [
    "monitor arm semihosting enable",
 ],
"showDevDebugOutput": true
```

I would have expected to see `monitor arm semihosting enable` somewhere in debug console:
```
0000000190: Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
0000000193: 1-gdb-set target-async on
0000000193: 2-interpreter-exec console "source /Users/andres/.vscode/extensions/marus25.cortex-debug-0.3.6/support/gdbsupport.init"
0000000193: 3-target-select extended-remote localhost:50000
0000000193: 4-interpreter-exec console "monitor reset halt"
0000000193: 5-target-download
0000000193: 6-interpreter-exec console "monitor reset halt"
0000000193: 7-enable-pretty-printing
```

haneefdm, 2020-05-07:
Yes, that is my github repo but it is old and for personal use when I was learning VSCode. Notice I was experimenting with multiple debuggers at the time, then decided this is a better extension for Cortex devices so I started contributing.

If you want it to add custom commands. with pre/post/override launch/attach/restart feel free.

You used post **restart** instead of **launch**.

andresv, 2020-05-07:
Thanks. This did the trick.

```
            "postLaunchCommands": [
                "monitor arm semihosting enable",
            ],
```